The-OpenROAD-Project/OpenROAD #3829: Render labels on top of the macro instead of to the left when zoomed out

oharboe, 2023-08-09:
### Description

open floorplan.odb in the GUI, zoom in/out.

zoomed out, the labels are to the left of the macros in some cases, but not all:

[image]

[floorplan_microwatt_sky130hd_base_2023-08-09_13-52.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12302250/floorplan_microwatt_sky130hd_base_2023-08-09_13-52.tar.gz)

gadfort, 2023-08-09:
@oharboe this is probably a result of your change for the centering.

oharboe, 2023-08-09:
> @oharboe this is probably a result of your change for the centering.

I need a bit of help of deciphering all the rotations, translations and coordinate systems there...

gadfort, 2023-08-09:
you are adding a positive xOffset in one case and negative in the other?
It also looks like the text is not really centered (slight below the center and slightly to the left of center).

gadfort, 2023-08-09:
@oharboe I don't know if I have time today to help

oharboe, 2023-08-09:
> @oharboe I don't know if I have time today to help

No rush for my part and I don't think anyone else either... I just wanted to get a reproducible test-case registered with this issue so it could be attended to in due time.

The current implementation works quite well.

I think it is struggling when the labels are long and the centering could be improved.

stefanottili, 2023-08-09:
I personally don’t think 90 degree rotated instance names are very readable and hope there are no upside down ones.
Luckily few technologies allow 90 degree rotations these days.

maliberty, 2023-08-09:
@stefanottili there are some macros that are tall and thin where rotation is the only useful orientation.  We could vary the threshold.

The label rotation is a function of the macro bounds and is never upside down or backwards.

PR #3832 (2023-08-11): gui: fix text rendering, recalculate dimensions after elide
We don't need great accuracy in choosing horizontal or vertical direction for text rendering, but once we have chosen direction, we want to center the text more precisely, so recalculate bounding box of text for elided text.

fixes #3829
commit 63b78af8 (2023-08-11): gui: fix text rendering, recalculate dimensions after elide

fixes #3829
closed 2023-08-11 by maliberty