The-OpenROAD-Project/OpenROAD #3838: openroad core dumps if 'read_def -child' is first read_def command

stefanottili, 2023-08-09:
### Describe the bug

# if -child is read first => coredump
read_def -child ./results/sky130hd/chameleon_hier_DFFRAM_4K/base/6_final.def
read_def ./results/sky130hd/chameleon/base/6_final.def

### Expected Behavior

No coredump. 
It works with the 'read_def -child' command after 'read_def <top>'.

Why is there a -child option needed ? 
Read multiple def files in any order and let the tool figure out which one is the top.
It's the one that doesn't get referenced, aka 'is not a component in any of the other def'.

### Environment

```shell
MacBook M1 Ventura
OpenROAD v2.0-9827-ge567c79e8
```

### To Reproduce

chameleon_hier.or
```
read_lef ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
read_lef ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
read_lef ./platforms/sky130ram/sky130_sram_1rw1r_128x256_8/sky130_sram_1rw1r_128x256_8.lef
read_lef ./platforms/sky130ram/sky130_sram_1rw1r_44x64_8/sky130_sram_1rw1r_44x64_8.lef
read_lef ./platforms/sky130ram/sky130_sram_1rw1r_64x256_8/sky130_sram_1rw1r_64x256_8.lef
read_lef ./platforms/sky130ram/sky130_sram_1rw1r_80x64_8/sky130_sram_1rw1r_80x64_8.lef
read_lef ./designs/sky130hd/chameleon/lef/apb_sys_0.lef
read_lef ./designs/sky130hd/chameleon/lef/DFFRAM_4K.lef
read_lef ./designs/sky130hd/chameleon/lef/DMC_32x16HC.lef
read_lef ./designs/sky130hd/chameleon/lef/ibex_wrapper.lef
# if -child is read first => coredump
read_def -child ./results/sky130hd/chameleon_hier_DFFRAM_4K/base/6_final.def
read_def ./results/sky130hd/chameleon/base/6_final.def
```
openroad -gui chameleon_hier.or

### Relevant log output

```shell
OpenROAD v2.0-9827-ge567c79e8 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd.tlef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0225]     Created 441 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130hd/lef/sky130_fd_sc_hd_merged.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_128x256_8/sky130_sram_1rw1r_128x256_8.lef
[INFO ODB-0221] 1000000 lines parsed!
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_128x256_8/sky130_sram_1rw1r_128x256_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_44x64_8/sky130_sram_1rw1r_44x64_8.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_44x64_8/sky130_sram_1rw1r_44x64_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_64x256_8/sky130_sram_1rw1r_64x256_8.lef
[INFO ODB-0221] 1000000 lines parsed!
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_64x256_8/sky130_sram_1rw1r_64x256_8.lef
[INFO ODB-0222] Reading LEF file: ./platforms/sky130ram/sky130_sram_1rw1r_80x64_8/sky130_sram_1rw1r_80x64_8.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./platforms/sky130ram/sky130_sram_1rw1r_80x64_8/sky130_sram_1rw1r_80x64_8.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/chameleon/lef/apb_sys_0.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./designs/sky130hd/chameleon/lef/apb_sys_0.lef at line 2.

[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/chameleon/lef/apb_sys_0.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/chameleon/lef/DFFRAM_4K.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./designs/sky130hd/chameleon/lef/DFFRAM_4K.lef at line 2.

[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/chameleon/lef/DFFRAM_4K.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/chameleon/lef/DMC_32x16HC.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./designs/sky130hd/chameleon/lef/DMC_32x16HC.lef at line 2.

[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/chameleon/lef/DMC_32x16HC.lef
[INFO ODB-0222] Reading LEF file: ./designs/sky130hd/chameleon/lef/ibex_wrapper.lef
[WARNING ODB-0220] WARNING (LEFPARS-2008): NOWIREEXTENSIONATPIN statement is obsolete in version 5.6 or later.
The NOWIREEXTENSIONATPIN statement will be ignored. See file ./designs/sky130hd/chameleon/lef/ibex_wrapper.lef at line 2.

[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  ./designs/sky130hd/chameleon/lef/ibex_wrapper.lef
Signal 11 received
Stack trace:
 0# handler(int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# ord::OpenRoad::readDef(char const*, odb::dbTech*, bool, bool, bool, bool) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 3# _wrap_read_def_cmd(void*, Tcl_Interp*, int, Tcl_Obj* const*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 4# TclNRRunCallbacks in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
 5# TclEvalEx in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
 6# Tcl_Eval in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
 7# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 8# gui::startGui(int&, char**, Tcl_Interp*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 9# ord::tclAppInit(Tcl_Interp*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
10# Tcl_MainEx in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
11# TclInitNamespaceSubsystem in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
12# main in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
zsh: segmentation fault  openroad -gui *.or
```

stefanottili, 2024-06-14:
Reading multiple def is currently only for the purpose of visualization.
Keep in mind that there might be more then two levels of def file hierarchy.

Link the def components after each read_def.
A) If there is a lef macro for every referenced def design, the order of def files doesn’t matter.
B) If there is no lef macro for a referenced def, require the def’s to be read “bottom up” aka, childs first. This way one can get the diearea and pin locations from def instead of a lef macro and/or create a lef macro “on the fly”.
This way there is no need for a -child option or an “all def read” command.

maliberty, 2024-06-17:
Even if you read bottom up you need to know when you are done and should display the result.  When reading Verilog we require `link_design` to indicate completion (as do other tools).   

If you allow reading out of order then you need to maintain intermediate data structures holding the unbound instances.  I think it simpler to keep the bottom up read order requirement to avoid that.  Once you agree to bottom up then using -child or adding a separate command become equivalent.