The-OpenROAD-Project/OpenROAD #3839: defRWarning.log only logs warnings of the last 'read_def -child'

stefanottili, 2023-08-10:
### Describe the bug

when reading multiple def files using 'read_def -child', warnings get written to defRWarning.log

1) But this file get's overwritten by each new read_def, so one only get's the warnings from the last file read.

2) [WARNING ODB-0149] DEF parser returns an error! but these don't show up in defRWarning.log

[WARNING ODB-0100] error: netlist component-pin (blockanme, pinname) is not defined
[WARNING ODB-0100] message limit reached, this message will no longer print
[WARNING ODB-0149] DEF parser returns an error!

some warnings caused by read_def get written to openroad -log log, some into defRWarning.log

Seems there are two different warnings, one from the parser, one from odb, so I assume this is "by design"

### Expected Behavior

multiple 'read_def -child' commands should append warnings to defRWarning.log instead of 'restarting it for every read_def'

### Environment

```shell
v2.0-9834-g36bd481ac
MacBook M1 Ventura
```

### To Reproduce

.

maliberty, 2023-08-10:
defRWarning.log is from the Si2 parser and we just left it on.  Is what is printed in OR a superset of what is in defRWarning.log?

stefanottili, 2023-08-10:
OR has a different set of warnings then defRWarning.log

defRWarning.log starts with the last <filename> read, so seems to "open to write" for each file,
overwriting the previous warnings. It would be good to "open to append".

Warnings from file, <filename>

annapetrosyan26, 2024-06-27:
Hello, could you please provide me the steps to reproduce this issue?

stefanottili, 2024-06-27:
Running flow/designs/asap7/mock-array you should get two defs, one for mock-array and one for mock-array Element.

Add some syntax errors to them and read both of them, one with the -child switch to get a hierarchical view.

annapetrosyan26, 2024-07-01:
Hello, with your provided steps I have created the following `mock_array.or` file which contains the following:
```
read_lef ./flow/platforms/asap7/lef/asap7_tech_1x_201209.lef
read_lef ./flow/platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
read_lef ./flow/results/asap7/mock-array_Element/base/Element.lef

read_def ./flow/results/asap7/mock-array/base/6_final.def
read_def -child ./flow/results/asap7/mock-array_Element/base/6_final.def
```
After running the `mock_array.or` with syntax errors in both def files I didn't get the `defRWarning.log` file as expected. Could you please advise if there might be something I'm missing or if there are additional steps I should take to reproduce the problem mentioned in this issue?

stefanottili, 2024-07-01:
You're right, I can't reproduce it using moc-array either.
Unfortunately I don't have access to the data I filed the bug report for any more.

If you look into the def parser sources, you see that there are some conditions under which defRWarning.log get's written to.
What I saw was that this was done "once per def file", so it was only reporting for the last read_def command.

grep -in defRWar ../tools/OpenROAD/src/odb/src/def/def/*
../tools/OpenROAD/src/odb/src/def/def/def.msg:156:7500 "Unable to open the file defRWarning.log in %s.\nWarning messages will not be written out in the log file.\nCheck if you have write permission on the directory."
../tools/OpenROAD/src/odb/src/def/def/def.msg:158:8500 "Unable to open the file defRWarning.log in %s.\nInfo messages will not be written out in the log file.\nCheck if you have write permission on the directory."
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1098:      if ((defrLog = fopen("defRWarning.log", "w")) == 0) {
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1100:            "WARNING(DEFPARS-8500): Unable to open the file defRWarning.log in "
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1115:      if ((defrLog = fopen("defRWarning.log", "a")) == 0) {
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1117:            "WARNING (DEFPARS-8500): Unable to open the file defRWarning.log "
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1178:      if ((defrLog = fopen("defRWarning.log", "w")) == 0) {
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1180:            "WARNING (DEFPARS-7500): Unable to open the file defRWarning.log "
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1195:      if ((defrLog = fopen("defRWarning.log", "a")) == 0) {
../tools/OpenROAD/src/odb/src/def/def/def_keywords.cpp:1197:            "WARNING (DEFAPRS-7501): Unable to open the file defRWarning.log "
../tools/OpenROAD/src/odb/src/def/def/defrReader.cpp:1113:  if (stat("defRWarning.log", &statbuf) != -1) {
../tools/OpenROAD/src/odb/src/def/def/defrReader.cpp:1116:      remove("defRWarning.log");
../tools/OpenROAD/src/odb/src/def/def/defrReader.hpp:695:// Routine to set the defrWarning.log to open as append instead for write

There is a routine to "append" to the log file ... but I wouldn't know where this get's called from.

grep -in defrSetOpenLogFileAppend ../tools/OpenROAD/src/odb/src/def/def/*
../tools/OpenROAD/src/odb/src/def/def/defrReader.cpp:2153:void defrSetOpenLogFileAppend()
../tools/OpenROAD/src/odb/src/def/def/defrReader.hpp:697:extern void defrSetOpenLogFileAppend();

razmikarshakyann, 2024-07-04:
Hi,
To generate the `defRWarning.log`, I called the `defWarning` function of `defrData`  manually inside the `defrRead` function, which is located in defrReader.cpp. In the `defWarning` function, there is the following code:
```
if ((defrLog = fopen("defRWarning.log", "w")) == 0) {
    printf(
            "WARNING (DEFPARS-7500): Unable to open the file defRWarning.log "
            "in %s.\n",
             getcwd(NULL, 64));
    printf("Warning messages will not be printed.\n");
 }
```
To append to the file instead of overwriting it, I just need to change "w" to "a". However, in the `defrRead` function, there is the following code:
```
if (stat("defRWarning.log", &statbuf) != -1) {
    /* file exist, remove it */
    if (!defContext.settings->LogFileAppend) {
      remove("defRWarning.log");
    }
  }
``` 
This code will always execute because `LogFileAppend` is set to 0 inside the `defrSettings` constructor. The `defContext.settings` are hard coded and there is no way to change these settings dynamically. To prevent log file removal, I can change `LogFileAppend`'s value to 1.

@stefanottili @maliberty  please let me know, if you would like me to create a pull request with this solution.