The-OpenROAD-Project/OpenROAD #3840: Add ability to drag the view with the mouse

oharboe, 2023-08-10:
It would make moving around the design a lot easier.

The arrow keys is a bit clunky compared to dragging the view with the mouse.

_Originally posted by @oharboe in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3812_

AcKoucher, 2023-08-29:
To be sure, drag should be triggered by shift+leftmouse?

oharboe, 2023-08-29:
> To be sure, dragging should be triggered by shift+leftmouse?

I don't know what recommendation to make here.... Something that isn't surprising for users of other EDA tools... Whatever that is...

For my part shift+leftmouse would work for me, if I could see or find in the GUI this combination...

maliberty, 2023-08-29:
in klayout it is the middle mouse button that does drag

PR #3925 (2023-09-14): gui: allow layout view to be dragged while holding middle mouse button
Resolves #3840

Layout can be dragged by holding middle mouse button like in KLayout.
closed 2023-09-14 by maliberty