The-OpenROAD-Project/OpenROAD #3845: AssignIroute_bestTrack Failure for I/O pins with TritinRoute

liangrj2014, 2023-08-11:
### Describe the bug

We generated a placed circuit (ariance133_68) at NanGate45 technology node, followed by global routing with cugr-2 ([cuhk-eda/cu-gr-2 (github.com)](https://github.com/cuhk-eda/cu-gr-2)). Then we tried to run TritinRoute on the global routed circuit but failed with "Error: assignIroute_bestTrack select no track for axi_req_o[261] @(0.0475, 643.37) (0.0475, 643.37 metal4 idx1/2=0/-". axi_req_o[261] is an I/O pin and we can successfully run TritinRoute on ariance133_68 by removing all I/O pins. 

### Expected Behavior

Resolve the assignIroute_bestTrack Failure for I/O pins with TritinRoute

### Environment

```shell
We installed the latest TritonRoute following the instructions on this link: https://github.com/The-OpenROAD-Project/TritonRoute

OS info:
NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.5 LTS"
VERSION_ID="20.04"
```

### To Reproduce

You can find related files in the following links:

https://drive.google.com/drive/folders/1Icbvzh3OsIsDxWGUToNLcYTObcYpGGrn?usp=sharing

DEF/LEF files are in ./benchmarks/NanGate45, routing guides generated by CUGR2 are in ./GR_solutions and log files output by TritinRoute are in ./DR_log.

### Relevant log output

```shell
via1 guide region query size = 0
metal2 guide region query size = 370216
via2 guide region query size = 0
metal3 guide region query size = 335166
via3 guide region query size = 0
metal4 guide region query size = 34062
via4 guide region query size = 0
metal5 guide region query size = 116809
via5 guide region query size = 0
metal6 guide region query size = 65556
via6 guide region query size = 0
metal7 guide region query size = 246
via7 guide region query size = 0
metal8 guide region query size = 197
via8 guide region query size = 0
metal9 guide region query size = 0
via9 guide region query size = 0
metal10 guide region query size = 0

init gr pin query ...
Waring: no output guide specified, skipped writing guide

start track assignment
Error: assignIroute_bestTrack select no track for axi_req_o[261] @(0.0475, 643.37) (0.0475, 643.37 metal4 idx1/2=0/-1
"TritonRoute_ariane133_68_ispd.log" 17437L, 760904C                                                              17437,117     Bot
```

maliberty, 2023-08-11:
@osamahammad21 please give this high priority

osamahammad21, 2023-08-12:
from DEF:

> TRACKS X 370 DO 4848 STEP 560 LAYER metal4 ;

vertical tracks on metal 4 start from 370. The guide is located before the first vertical track on metal4. Thus no track is available to be assigned for that guide.
[image]
P.S. From the relevant log, I see that you are using the standalone TritonRoute which we don't support anymore. Please use the router module through openroad.

maliberty, 2023-08-12:
How were the guides generated?

osamahammad21, 2023-08-14:
@maliberty According to the issue description, they used cugr2.
In my opinion there is nothing wrong with the detailed router in this case so far.

maliberty, 2023-08-14:
In that case it is a cugr-2 problem.  Is there any reason to not use OR's global router?

liangrj2014, 2023-08-14:
Thanks a lot for all the helps! Yes, the guides are generated with cugr2. We will try OR's global router or fix the bugs in the guides generated by cugr2. And we will use the router module in OR. Really appreciate your feedbacks!

closed 2023-08-14 by maliberty