The-OpenROAD-Project/OpenROAD #3847: OR coredump triggered by mock-array_Element save_clocktree_image

stefanottili, 2023-08-11:
### Describe the bug

openroad coredumps in save_clocktree_image: see logs/asap7/mock-array_Element/base/6_report.log below

### Expected Behavior

no coredump

### Environment

```shell
MacBook M1 Ventura, mpl2 compiled in
v2.0-9837-gb31a3b960
```

### To Reproduce

make DESIGN_CONFIG=./designs/asap7/mock-array/config.mk

### Relevant log output

```shell
logs/asap7/mock-array_Element/base/6_report.log

558 ==========================================================================
559 finish report_design_area
560 --------------------------------------------------------------------------
561 Design area 203 u^2 12% utilization.
562 
563 [WARNING GUI-0076] Populating font family aliases took 48 ms. Replace uses of missing font family "Monospace" with one that exists to avoid this cost.
564 Signal 11 received
565 Stack trace:
566  0# handler(int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
567  1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
568  2# sta::dbNetwork::term(sta::Pin const*) const in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
569  3# gui::ClockTree::ClockTree(sta::Clock*, sta::dbNetwork*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
570  4# gui::STAGuiInterface::getClockTrees() const in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
571  5# gui::ClockWidget::populate(sta::Corner*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
572  6# gui::ClockWidget::saveImage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::ch
573  7# gui::Gui::saveClockTreeImage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::c
574  8# save_clocktree_image(char const*, char const*, char const*, int, int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
575  9# _wrap_save_clocktree_image(void*, Tcl_Interp*, int, Tcl_Obj* const*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
576 10# TclNRRunCallbacks in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
577 11# TclEvalEx in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
578 12# Tcl_Eval in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
579 13# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
580 14# gui::startGui(int&, char**, Tcl_Interp*, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, bool) in /Users/user/OpenROA
581 15# show(char const*, bool) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
582 16# _wrap_show(void*, Tcl_Interp*, int, Tcl_Obj* const*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
583 17# TclNRRunCallbacks in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
584 18# TclEvalEx in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
585 19# Tcl_Eval in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
586 20# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
587 21# ord::tclAppInit(Tcl_Interp*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
588 22# Tcl_MainEx in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
589 23# TclInitNamespaceSubsystem in /opt/homebrew/Cellar/tcl-tk/8.6.13_4/lib/libtcl8.6.dylib
590 24# main in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
```

stefanottili, 2023-08-11:
commenting out the save_images.tcl in ./scripts/final_report.tcl gets me past the coredump
67 #    gui::show "source $::env(SCRIPTS_DIR)/save_images.tcl" false

```
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
[INFO][FLOW] Invoked hierarchical flow.
Block Element needs to be hardened.
Makefile:409: warning: overriding commands for target `&'
Makefile:322: warning: ignoring old commands for target `&'
/Library/Developer/CommandLineTools/usr/bin/make DESIGN_CONFIG=./designs/asap7/mock-array/config.mk DESIGN_CONFIG=./designs/asap7/mock-array/Element/config.mk generate_abstract
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
(/usr/bin/time stdbuf -o L /Users/user/bin/klayout -zz -rd design_name=Element \
                -rd in_def=results/asap7/mock-array_Element/base/6_final.def \
                -rd in_files="./platforms/asap7/gds/asap7sc7p5t_28_R_220121a.gds  " \
                -rd config_file= \
                -rd seal_file="" \
                -rd out_file=results/asap7/mock-array_Element/base/6_1_merged.gds \
                -rd tech_file=./objects/asap7/mock-array_Element/base/klayout.lyt \
                -rd layer_map= \
                -r ./util/def2stream.py) 2>&1 | tee ./logs/asap7/mock-array_Element/base/6_1_merge.log
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
ERROR: RuntimeError: Read error on file: /Users/user/OpenROAD-flow-scripts/flow/results/asap7/mock-array_Element/base (errno=21) in Layout.read
  ./util/def2stream.py:113
        0.22 real         0.18 user         0.02 sys
cp results/asap7/mock-array_Element/base/6_1_merged.gds results/asap7/mock-array_Element/base/6_final.gds
cp: results/asap7/mock-array_Element/base/6_1_merged.gds: No such file or directory
```

maliberty, 2023-08-11:
This must be a mac specific issue as it passes in the linux CI.  @eder-matheus do you see this?

stefanottili, 2023-08-12:
The def2gds issue disappeared after blowing away everything inside the flow directory and doing a fresh git checkout.

The coredump is still there.

stefanottili, 2023-08-14:
You can't deduce that by having a passing CI, you don't have this coredump.

The flow continues regardless of the coredump, since it's only saving an image.
And the logs don't capture the coredump at all.

So this might coredump on other machines/OS'es too.

maliberty, 2023-08-14:
If the tool coredumps then it should have a non-zero exit and make should detect that step as failed.  Do you not see that?

maliberty, 2023-08-15:
we only build on mac, we don't have the resources to run the flow

stefanottili, 2023-10-10:
just rechecking after all the recent sta updates ... this coredump still exists.
v2.0-10644-g3187676d8

maliberty, 2023-10-10:
It just passed in the CI: https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/3112/execution/node/701/log/

maliberty, 2023-10-10:
Same commit OpenROAD v2.0-10644-g3187676d8

stefanottili, 2023-10-10:
the relevant log is:

https://content.jenkins.openroad.tools/static-files/s80IrI89B6Tt56y40xZ9W-YTtRldrmfgTKKhjV-r_HoxNjk2OTY1Mjk1NTI5Ojk6YW5vbnltb3VzOmpvYi9PcGVuUk9BRC1mbG93LXNjcmlwdHMtTmlnaHRseS1QdWJsaWMvMzExMi9hcnRpZmFjdA==/flow/logs/asap7/mock-array_Element/base/6_report.log

So if it passes CI, what makes it coredump using apple clang on macOS ?

It's core dumping in the last step that generates 6_report.log, scripts/save_images.tcl for me.

77 foreach clock [get_clocks *] {
 78   if { [llength [get_property $clock sources]] > 0 } {
 79     set clock_name [get_name $clock]
 80     gui::save_clocktree_image $::env(REPORTS_DIR)/cts_$clock_name.webp $clock_name
 81   }
 82 }

Does it create clock tree images in CI or when you run it locally ?

Any suggestion on how to debug this ? 
Either how to run this openroad "stand alone" in a debugger or where to put some debug printf's ?

 0# handler(int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 1# _sigtramp in /usr/lib/system/libsystem_platform.dylib
 2# sta::dbNetwork::term(sta::Pin const*) const in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 3# gui::ClockTree::ClockTree(sta::Clock*, sta::dbNetwork*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 4# gui::STAGuiInterface::getClockTrees() const in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 5# gui::ClockWidget::populate(sta::Corner*) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 6# gui::ClockWidget::saveImage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::optional<int> const&, std::__1::optional<int> const&) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 7# gui::Gui::saveClockTreeImage(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, int, int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad
 8# save_clocktree_image(char const*, char const*, char const*, int, int) in /Users/user/OpenROAD-flow-scripts/tools/install/OpenRoad/bin/openroad

maliberty, 2023-10-10:
It saves the images in the CI.  I would do 'make fnal_report_issue' which will generate the scripts to run that one step.

maliberty, 2023-10-17:
I don't see any path to resolve this mac specific issue.

closed 2023-10-17 by maliberty

stefanottili, 2023-10-18:
Even with the sta updates to v2.0-10744-g8ed045098, this still consistently and reproduceably coredumps running "safe_clocktree_image" inside sta::dbNetwork::term for both moc-array and mock-array-Element as seen in 6_report.log.

I didn't debug it further because the "make final_report_issue" generated shell script has bugs that would warrant another bug report and I ran out of time.

I don't see why this issue has been be closed, even when it doesn't occur on linux, it's still a bug.

It might warrant a valgrind or address-sanitizer run of this step.

maliberty, 2023-10-18:
If you plan to address the issue we can reopen it.  If not there is no point in having an open issue that nobody is going to address.