The-OpenROAD-Project/OpenROAD #3859: psm error in asap7/riscv32i

maliberty, 2023-08-15:
### Describe the bug

The CI is now showing:

[image]

I suspect this relates to your recent PR

### Expected Behavior

No error

### Environment

```shell
https://jenkins.openroad.tools/blue/organizations/jenkins/OpenROAD-flow-scripts-Nightly-Public/detail/OpenROAD-flow-scripts-Nightly-Public/2880/pipeline/729
```

### To Reproduce

See https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/2880/artifact/flow/final_report_riscv32i_asap7_base_2023-08-14_19-26.tar.gz

gadfort, 2023-08-15:
@maliberty The check is correct (although there is a fix I need to do there, I'll submit that tomorrow).
It looks like the two of the four memories are not getting connected to the power grid (the previous code would have "connected" the macros on M6 despite there being no vias connecting the iterms of the memories to the grid).

So three actions needed:
1. check_power_grid and analyze_power_grid differ in their error mode (check_power_grid doesn't see this error, but analyze_power_grid does, this should not be the case). (I'll fix this check tomorrow)
2. pdngen should have errored out when it failed to connect the macros to anything (I'll fix this later this week)
3. the asap7 power grid isn't capable to connecting to these memories (your team will need to correct this)

maliberty, 2023-08-15:
Is it not capable of connecting because of the narrow width of the RAMs?

gadfort, 2023-08-15:
@maliberty fixing check_power_grid is going to take a little longer, it seems to have opened pandoras box of shortcuts.
The reason the RAMs are not getting connected is they don't have a real power grid defined, just a single set of connect statements but without straps over them on those layers, there is nothing to connect (only one memory is getting connected by chance really)

maliberty, 2023-08-15:
In asap7 I see
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro -orient {R0 R180 MX MY} -cells {.*}
```
vs 
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro -orient {R0 R180 MX MY} -halo {2.0 2.0 2.0 2.0} -default -grid_over_boundary
```
in sky130hd.  Is the -cells vs "-default -grid_over_boundary" the issue?

gadfort, 2023-08-16:
@maliberty no that doesn't matter, the m5 straps just don't intersect the memories.

maliberty, 2023-08-16:
So if the macro width is less than the strap pitch then pdn will just fail to connect?  Isn't it akin to a repair strap that you just need to add an extra for the macro?

gadfort, 2023-08-16:
I suppose it is, but thats not what the repair channel code is currently capable of doing. Secondly, just adding straps might get you connectivity, but you may not meet any kind of IR drop.  I think PDNGEN should have errored out and I'll add that check to it so we don't have incomplete power grids, but I don't have time to look at the repair channel thing.

maliberty, 2023-08-17:
Resolved by https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1371

closed 2023-08-17 by maliberty