The-OpenROAD-Project/OpenROAD #3861: FlexDRWorker::getTrackLocs needs implementation fix

osamahammad21, 2023-08-15:
### Description

https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a66a6b07a3231b9400ed35d0f4e71eace5f5f4f/src/drt/src/dr/FlexDR_init.cpp#L1394-L1401

trackLocs is a set. At the first line, you already add the trackLoc to the set. The if condition after that is completely useless. And this is the only part where the bool isHorzTracks is used, so it is completely useless as well.

### Suggested Solution

Check the usage of this function and the expected results to determine how to fix.

refaay, 2023-09-21:
`https://github.com/The-OpenROAD-Project/OpenROAD/blob/6a66a6b07a3231b9400ed35d0f4e71eace5f5f4f/src/drt/src/dr/FlexDR_init.cpp#L1102-L1107`
While reviewing the code, I also found this weird if condition that will never be entered because "useCenterLine" is always true at this point.

rovinski, 2023-09-25:
> So my question is: is having two consecutive layers of the same preferred direction valid?

I have never seen a technology do this. In general, it wouldn't make sense. I think it would be fine to instead error on this condition and not support it unless someone finds a genuine need.