The-OpenROAD-Project/OpenROAD #3866: Improve mpl2 so it can lay out unabutted macros in mock-array

oharboe, 2023-08-16:
### Description

This is the desired floorplan:

[image]

Make the changes below to set up mpl2 for unabutted floorplan:

```
$ git diff
diff --git a/flow/designs/asap7/mock-array/config.mk b/flow/designs/asap7/mock-array/config.mk
index a2a30e83..a884177a 100644
--- a/flow/designs/asap7/mock-array/config.mk
+++ b/flow/designs/asap7/mock-array/config.mk
@@ -28,6 +28,8 @@ BLOCKS                       = Element
 
 export GDS_ALLOW_EMPTY       = Element
 
+#export SYNTH_HIERARCHICAL = 1
+#export RTLMP_FLOW = True
 export MACRO_PLACEMENT_TCL   = ./designs/asap7/mock-array/macro-placement.tcl
 
 export IO_CONSTRAINTS        = designs/asap7/mock-array/io.tcl
diff --git a/flow/designs/asap7/mock-array/defaults.mk b/flow/designs/asap7/mock-array/defaults.mk
index bd8eb304..959d0761 100644
--- a/flow/designs/asap7/mock-array/defaults.mk
+++ b/flow/designs/asap7/mock-array/defaults.mk
@@ -3,7 +3,7 @@
 # current unit is configured as 2.16 which is on the routing grid for M5
 
 # table of Elements - (rows cols width height pitch_x pitch_y)
-export MOCK_ARRAY_TABLE         ?= 8 8 20 20 20 22
+export MOCK_ARRAY_TABLE         ?= 8 8 20 20 22 22
 
 # Element'd data width
 export MOCK_ARRAY_DATAWIDTH     ?= 64
```

Then run:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk
```

Result:

```
[deleted]

+---R1  (8)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---R2  (9)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---B0  (10)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---B1  (11)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---B2  (12)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

The design only has macros.

[Hier-RTLMP::HardMacroClusterMacroPlacement] Place macros in cluster: root
[ERROR MPL-0010] Cannot find valid macro placement for hard macro cluster: root
Error: macro_place.tcl, 141 MPL-0010
Command exited with non-zero status 1
Elapsed time: 0:07.57[h:]min:sec. CPU time: user 58.93 sys 0.04 (778%). Peak memory: 234884KB.
make[1]: *** [Makefile:522: do-2_4_floorplan_macro] Error 1
make[1]: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow'
make: *** [Makefile:522: results/asap7/mock-array/base/2_4_floorplan_macro.odb] Error 2

```

oharboe, 2023-08-16:
Alternative way to reproduce with `make macro_place_issue`:

[mpl2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12355316/mpl2.zip)

+---B0  (10)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---B1  (11)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

+---B2  (12)  num_macro :  0   num_std_cell :  0  macro_area :  0.0  std_cell_area : 0.0

The design only has macros.

```
./run-me-mock-array-asap7-base.sh
[deleted]
[Hier-RTLMP::HardMacroClusterMacroPlacement] Place macros in cluster: root
[ERROR MPL-0010] Cannot find valid macro placement for hard macro cluster: root
Error: macro_place.tcl, 141 MPL-0010
openroad> 
```

maliberty, 2023-08-16:
You have: export MACRO_PLACE_HALO="10 10"
and the element is: SIZE 43.2 BY 43.2 ;
the block is 388.8 x 388.8

There is no way to make this work.  10 + (43.2 + 10) * 8 = 435.6

oharboe, 2023-08-16:
> You have: export MACRO_PLACE_HALO="10 10" and the element is: SIZE 43.2 BY 43.2 ; the block is 388.8 x 388.8
> 
> There is no way to make this work. 10 + (43.2 + 10) * 8 = 435.6

What should I set MACRO_PLACE_HALO to in this case?

maliberty, 2023-08-16:
What do you expect the macro spacing to be in your desired floorplan?

oharboe, 2023-08-16:
> What do you expect the macro spacing to be in your desired floorplan?

I was just going to pick some distance where we expect mpl2 to work at this point.

maliberty, 2023-08-16:
You have to make the floorplan large enough to accommodate what you pick.   mpl2 will not change the block size.

QuantamHD, 2023-08-16:
> You have to make the floorplan large enough to accommodate what you pick. mpl2 will not change the block size.

Why not? Seems helpful

maliberty, 2023-08-16:
> > You have to make the floorplan large enough to accommodate what you pick. mpl2 will not change the block size.
> 
> Why not? Seems helpful

Row are already set, IOs already placed.  Nobody would expect it and it overrides a specific user input.  Would break top level floorplanning.  etc.

oharboe, 2023-08-16:
changed to a feature request to make mpl2 more user-friendly.

maliberty, 2023-08-16:
What would make it more friendly?

oharboe, 2023-08-16:
> What would make it more friendly?

No need to tinker with MACRO_PLACE_HALO or understand what it does to get a result unless you want to tweak what you get.

There are multiple halo and channel variables if one just wants the macros placed to focus on something else, then it is nice that one gets a result without having to learn about the finer details.

maliberty, 2023-08-16:
How should we know when to obey a user input and when to just ignore it?

oharboe, 2023-08-16:
Just document that that is the policy of mpl2. mpl2 has great freedom in how macros are placed, this is a little bit more freedom.

maliberty, 2023-08-16:
It has freedom within the given inputs.  It doesn't have freedom to ignore those inputs.  This is true of every command.  What needs documenting?  I'm confused.

oharboe, 2023-08-17:
I have thought more about what you said concerning using the variables to strictly constrain mpl2(and other parts of the system).

That can be user-friendly if mpl2 prints out clean actionable error messages.

Placement will tell you what PLACE_DENSITY value to try next. That works very well.

The same for mpl2 and halo and floorplan area (and whatever other parameters there are), would address my concerns in this feature request.

maliberty, 2023-08-17:
As an example imagine a 1x5 macro and a 5x1 macro.  You can't place them in 10 units of area.

oharboe, 2023-08-17:
If mpl2 could recommend a new halo number, then that would be an improvement. Similarly, it could recommend increasing the area by some percentage. This would guide users and make mpl2 more friendly.

maliberty, 2023-08-17:
I don't see a way to give a good suggestion.  It might be possible in your simple case but it isn't in general as I noted above.  If you have an algorithm that works in general please share it.

maliberty, 2023-08-17:
I'll add
```
[ERROR MPL-0016] The instance area with halos 181158.73 exceeds the core area 147456.0
```

oharboe, 2023-08-17:
> I'll add
> 
> ```
> [ERROR MPL-0016] The instance area with halos 181158.73 exceeds the core area 147456.0
> ```

It sounds like you have the halo area, so add those numbers to error message?

(just example numbers)

```
[ERROR MPL-0016] The instance area (12342.0) with halos(1221.0), 181158.73 exceeds the core area 147456.0
```

maliberty, 2023-08-17:
Right before the error we have already printed
```
	Number of std cell instances : 130
	Area of std cell instances : 23.44
	Number of macros : 64
	Area of macros : 119439.39
	Area of macros with halos : 181135.30
	Total area : 119462.84
	Design Utilization : 0.81
	Core Utilization: 0.00
	Manufacturing Grid: 1
```

oharboe, 2023-08-17:
I see. I think your error message is a clear improvement.

Let's try that and mark this one as closed pending some new information on how more accurate and actionable error messages could be articulated. We're not holding our breath for a better algorithm...

oharboe, 2023-08-18:
Work pretty well. Thanks!

closed 2023-08-18 by oharboe