The-OpenROAD-Project/OpenROAD #3875: improve mpl2 so that it lays out mock-array efficiently

oharboe, 2023-08-18:
### Description

With this example, [mpl2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12376554/mpl2.zip), mpl2 achieves the following layout, which is fine, except the elements are jumbled.

The bottom row starts ideally laid out ces_0_0 .. ces_0_7, the second row from the bottom should be  ces_1_0 .. ces_1_7, but starts withces_2_0

[image]

To produce the archive above, I ran `make macro_place_issue` with the following changes:

```
diff --git a/flow/designs/asap7/mock-array/config.mk b/flow/designs/asap7/mock-array/config.mk
index a2a30e83..a33724d9 100644
--- a/flow/designs/asap7/mock-array/config.mk
+++ b/flow/designs/asap7/mock-array/config.mk
@@ -28,7 +28,10 @@ BLOCKS                       = Element
 
 export GDS_ALLOW_EMPTY       = Element
 
-export MACRO_PLACEMENT_TCL   = ./designs/asap7/mock-array/macro-placement.tcl
+#export MACRO_PLACEMENT_TCL   = ./designs/asap7/mock-array/macro-placement.tcl
+export SYNTH_HIERARCHICAL = 1
+export RTLMP_FLOW = True
+#export MACRO_PLACE_HALO = 0.5 0.5
 
 export IO_CONSTRAINTS        = designs/asap7/mock-array/io.tcl
 
diff --git a/flow/designs/asap7/mock-array/defaults.mk b/flow/designs/asap7/mock-array/defaults.mk
index bd8eb304..579b3a3d 100644
--- a/flow/designs/asap7/mock-array/defaults.mk
+++ b/flow/designs/asap7/mock-array/defaults.mk
@@ -3,7 +3,7 @@
 # current unit is configured as 2.16 which is on the routing grid for M5
 
 # table of Elements - (rows cols width height pitch_x pitch_y)
-export MOCK_ARRAY_TABLE         ?= 8 8 20 20 20 22
+export MOCK_ARRAY_TABLE         ?= 8 8 20 20 30 30
 
 # Element'd data width
 export MOCK_ARRAY_DATAWIDTH     ?= 64
```

### Suggested Solution

mpl2 should be able to figure out the best layout for this array.

maliberty, 2023-08-21:
There are 64! possible arrangements and only one acceptable answer.  We can run more iterations of annealing but I expect it will take a long time to find the answer you expect.

What do you mean by "RTL ordering"?  Are you referring to name based sorting?

oharboe, 2023-08-22:
> There are 64! possible arrangements and only one acceptable answer. We can run more iterations of annealing but I expect it will take a long time to find the answer you expect.

Indeed. There needa to be a way to eliminate equally good/bad solutions without testing.
> 
> What do you mean by "RTL ordering"? Are you referring to name based sorting?

The Verilog has an ordering. Use it to create a preference when there is nothing else to distinguish.

maliberty, 2023-08-22:
In what way does the verilog have an ordering?

oharboe, 2023-08-22:
> In what way does the verilog have an ordering?

If you read the RTL top to bottom, you find the theock array compute elements ordered in the right array order.

maliberty, 2023-08-22:
That's not a general property. I think we need a way to pass a hint/constraint to mpl.

oharboe, 2023-08-23:
> That's not a general property. I think we need a way to pass a hint/constraint to mpl.

Yes, some constraints would work.

I can grep to create a list of macro names and feed that in as a constraing.

I dont need or want rotation.

oharboe, 2023-08-23:
> Rotation is much easier to get right but the current code isn't that smart. I think that can be fixed more easily.

Sounds good. The fewer options the better if it gets the job done....

maliberty, 2023-10-03:
Partially addressed in #3973 with another PR to follow (in testing)

maliberty, 2023-10-04:
With these two PRs the design is laid out as expected and routing completes in 3 iterations.

closed 2023-10-04 by maliberty

oharboe, 2023-10-04:
It tried this with a slightly different test-case after this issue was closed and filed an additional feature request #4085