The-OpenROAD-Project/OpenROAD #3878: Add meaningful information to resolve the issue

vijayank88, 2023-08-19:
### Description

https://openroad.readthedocs.io/en/latest/user/MessagesFinal.html 
It is great to trace currently the source code of `ERROR/WARN`.

Based on current flow, it is good to add possible way to fix the ERROR by changing which variables or which stage of the flow need to be checked to resolve that particular issue. It is useful those who have less experience with source code files.

While running riscv32i, by changing some default variables got below error with ORFS flow.
```
[ERROR DPL-0044] Cell dmem.dmem0 with height 42000 is taller than any row.
Error: detail_place.tcl, 10 DPL-0044
```
Here what are the variables I need to tweak to resolve the error.

1. Placement density
2. Cell Padding
3. Re-Floorplan
Nothing helps to fix the issue, then user can file a github issue to resolve the issue.

### Suggested Solution

Update https://openroad.readthedocs.io/en/latest/user/MessagesFinal.html `Information` with meaningful info.

maliberty, 2023-08-19:
What setting did you tweak?  This is clearly a memory and should be placed by the macro placer or a user and should never be handled by dpl.

QuantamHD, 2023-08-21:
Maybe we should include a warning or better error message?

vijayank88, 2023-08-21:
@maliberty 
Just commented lines https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/designs/asap7/riscv32i/config.mk#L5-L7 to skip RTLMP and use MPL for macro placement.
Then planning to use manual macro placement for lab exercise.

maliberty, 2023-08-21:
> Maybe we should include a warning or better error message?

What do you have in mind?

maliberty, 2024-03-05:
Stale

closed 2024-03-05 by maliberty