The-OpenROAD-Project/OpenROAD #3883: No Access Point for cell during Detailed routing for TSMC65 PDK [ERROR DRT-0073] OpenRoad

faisaladilquadery, 2023-08-21:
### Describe the bug

During the insertion of pdk TSMC 65 in openroad, this following error shows during the detailed route stage. I have reduced the congestion of the design to make sure it is not a congestion issue to 20 percent utilization.
[image]

Some Examples of the instances are shown here : 
[image]

[image]

[image]

### Expected Behavior

Expectation is the design to complete detailed routing. 
If I manually dont use the particular cell with no access, the design completes the routing. 

### Environment

```shell
tee: build/openroad-env.log: Permission denied
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-46-generic
os: Ubuntu 22.04 (Jammy Jellyfish)
cmake version 3.24.2
CMake Error at CMakeLists.txt:135 (configure_file):
  Operation not permitted

-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9070-gbb341ffb9
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 87 lines cut ...]
-- Found re2: /opt/or-tools/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring incomplete, errors occurred!
```

### To Reproduce

./.

maliberty, 2023-08-21:
Access point problems don't relate to congestion.  They are about how to drop a set of vias onto the pins without creating DRCs between them.  Without a test case they are hard to debug.

I suggest you try the developer debug with
detailed_route_debug -pa -pa_markers -pin input22:Z

and run the router in the GUI.  It should stop on that pin and show you the various things it is trying and any DRCs they generate.  That can give a clue as to what is going on.

Where are you getting your std cell library from?

faisaladilquadery, 2023-08-21:
Hello maliberty, 
Thankyou for the feedback, I will try it and let you know.
I am using the TSMC 65 pdk and trying it to integrate into the flow.

faisaladilquadery, 2023-08-21:
I have used these commands and got the following response is shown in screenshots. It is showing a minimum cut, unkown length for the pin. Can you suggest the solution ?

>>> detailed_route_debug -pa -pa_markers -pin input22:Z
>>> detailed_route -bottom_routing_layer M1 -top_routing_layer M7 -save_guide_updates -verbose 1

[image]
[image]
[image]

maliberty, 2023-08-21:
It looks like this cell has a pin with shapes large enough to require a double cut via.  OR doesn't support that so I suggest you mark this cell as dont_use and avoid it.

I think you should be careful about what you post here as I assume you have an NDA around this data.  I've removed your post above out of caution.

faisaladilquadery, 2023-08-21:
Thankyou so much for your response. Yes it was careless of me, you can remove the post.

closed 2023-08-21 by faisaladilquadery