The-OpenROAD-Project/OpenROAD #3891: Voltage on net VDD is not set

kareefardi, 2023-08-22:
### Describe the bug

gf180mcuC flow in OpenLane is quitting during `analyze_power_grid`. This is new behavior.

### Expected Behavior

Voltage is not explicitly set. However we didn't run into this issue in sky130 flow.

### OpenROAD Environment

```shell
Git commit: 289efb0beb9204d52cc29e9dc900997dbae039c1
kernel: Linux 5.15.0-78-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9957-g289efb0be
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 119 lines cut ...]
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

Extract and `bash runs.sh`

### Relevant log output

```shell
define_corners Typical
read_liberty -corner Typical pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
Using 1e-12 for capacitance...
Using 1e+00 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-06 for power...
Using 1e-06 for distance...
Reading design constraints file at './tmp/17-spm.sdc'…
[INFO]: Setting RC values...
[ERROR PSM-0093] Voltage on net VDD is not set.
Error: irdrop.tcl, 35 PSM-0093
```

donn, 2023-08-22:
Additionally, I should note that this is a regression- for whatever reason, in 0cfb9a45bfb256c9af1a0500d4c97da0f145f54f, this works fine.

Now, we always favor the side of being more explicit, but it'd be nice to know what's absent in gf180mcu that isn't in sky130.

vijayank88, 2023-08-25:
@kareefardi 
How to reproduce this?

kareefardi, 2023-08-25:
@vijayank88 I forgot to upload. Thanks for the headsup. I will edit the main comment.

vijayank88, 2023-08-25:
@kareefardi 
seems you missed to define `set_pdnsim_net_voltage` before `analyze_power_grid`.
Check ORFS script [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/final_report.tcl#L41C1-L41C1).

kareefardi, 2023-08-25:
I see. But as @donn was saying. Is there a reason why it works with sky130 even without `set_pdnsim_net_voltage`. See reproducible:
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12438483/issue_reproducible.tar.gz)

vijayank88, 2023-08-25:
@gadfort 
Can you have a look into this?

maliberty, 2023-08-25:
In sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib:
```
    default_operating_conditions : "tt_025C_1v80";
    operating_conditions ("tt_025C_1v80") {
        voltage : 1.8000000000;
        process : 1.0000000000;
        temperature : 25.000000000;
        tree_type : "balanced_tree";
    }
```
so we can take the voltage from there.  In gf180 there is no default_operating_conditions so we can't.

closed 2023-08-25 by maliberty