The-OpenROAD-Project/OpenROAD #3892: Make antenna checker understand global route format

eder-matheus, 2023-08-22:
### Description

When evaluating the global routing results for antenna violations, we translate the global route format (i.e., global segments) to actual wires, having to create local connections with the pins.

This translation sometimes has issues, leading to crashes, specifically on the `odb::orderWires` function.

### Suggested Solution

The antenna checker tool should be able to understand the global route format, removing the necessity of translating the global route to wires. This would remove a large piece of code that performs the translation, remove the calls for `odb::orderWires`, and basically would simplify the process of checking antenna violations on the global route results.

### Additional Context

I'm not 100% sure if supporting the global route format in the antenna checker is possible. Any input on potential problems this could lead will be appreciated.

maliberty, 2023-08-22:
I think this is possible. Antenna checking just needs a topology and metal area attached.

openroadie, 2023-08-24:
@maliberty : We discussed it in the weekly meeting and I was going to take a look since @eder-matheus has some other higher priority items on his list.

maliberty, 2024-03-13:
In https://github.com/The-OpenROAD-Project/OpenROAD/pull/4673 I've accepted some degradation in sky130hs/aes that originates from a new detour leading to more antenna fixing.  Once the antenna fixing is improved please see if this design is improved.  Currently there are some odd antennas on the driver side.

closed 2024-06-05 by vvbandeira