The-OpenROAD-Project/OpenROAD #3907: gate closing leaves floating nets behind

gadfort, 2023-08-26:
### Describe the bug

When running `repair_timing` with the new gate cloning logic floating nets appear in the design indicating that there are errors occurring.

```
[WARNING RSZ-0020] found 79 floating nets.
```

### Expected Behavior

No changes to the functionality of the design and no floating nets that weren't there before.

### Environment

```shell
OpenROAD v2.0-10001-g53a32a198
```

### To Reproduce

[sc_issue_aes_job0_freepdk45_demo_orfs_cts0_20230825-230053.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12446011/sc_issue_aes_job0_freepdk45_demo_orfs_cts0_20230825-230053.tar.gz)

```
tar xvf sc_issue_aes_job0_freepdk45_demo_orfs_cts0_20230825-230053.tar.gz
cd sc_issue_aes_job0_freepdk45_demo_orfs_cts0_20230825-230053
./run.sh
```

### Relevant log output

```shell
OpenROAD v2.0-10001-g53a32a198 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO FLW-0001] Defining timing corners: typical
Reading liberty file for typical: /home/pgadfort/siliconcompiler/third_party/pdks/virtual/freepdk45/libs/nangate45/r1p0/lib/NangateOpenCellLibrary_typical.lib
Reading ODB: inputs/aes_cipher_top.odb
Reading SDC: inputs/aes_cipher_top.sdc
[INFO FLW-0001] Using metal5 for clock parasitics estimation
[INFO FLW-0001] Using metal3 for signal parasitics estimation
[INFO ORD-0030] Using 16 thread(s).
[INFO FLW-0001] Setting global routing adjustment for metal1 to 100.0%
[INFO FLW-0001] Setting global routing adjustment for metal2 to 50.0%
[INFO FLW-0001] Setting global routing adjustment for metal3 to 50.0%
[INFO FLW-0001] Setting global routing adjustment for metal4 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal5 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal6 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal7 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal8 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal9 to 25.0%
[INFO FLW-0001] Setting global routing adjustment for metal10 to 25.0%
[INFO FLW-0001] Setting global routing signal routing layers to: metal2-metal10
[INFO FLW-0001] Setting global routing clock routing layers to: metal2-metal10
#######################################
# Clock tree synthesis
# (skip if no clocks defined)
#######################################
if {[llength [all_clocks]] > 0} {
  # Clone clock tree inverters next to register loads
  # so cts does not try to buffer the inverted clocks.
  repair_clock_inverters
  set sc_cts_arguments []
  if {$openroad_cts_balance_levels == "true"} {
    lappend sc_cts_arguments "-balance_levels"
  }
  clock_tree_synthesis -root_buf $sc_clkbuf -buf_list $sc_clkbuf \
    -sink_clustering_enable \
    -sink_clustering_size $openroad_cts_cluster_size \
    -sink_clustering_max_diameter $openroad_cts_cluster_diameter \
    -distance_between_buffers $openroad_cts_distance_between_buffers \
[... 824 lines cut ...]
SC_METRIC: fmax
960.0213894408657 MHz
SC_METRIC: power
Power for corner: typical
Group                  Internal  Switching    Leakage      Total
                          Power      Power      Power      Power (Watts)
----------------------------------------------------------------
Sequential             1.16e-02   4.77e-03   5.90e-05   1.64e-02   4.0%
Combinational          1.93e-01   2.01e-01   5.45e-04   3.95e-01  96.0%
Macro                  0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
Pad                    0.00e+00   0.00e+00   0.00e+00   0.00e+00   0.0%
----------------------------------------------------------------
Total                  2.05e-01   2.06e-01   6.04e-04   4.11e-01 100.0%
                          49.8%      50.0%       0.1%

SC_METRIC: cellarea
Design area 23343 u^2 12% utilization.
[INFO FLW-0001] Saving "snapshot" to reports/images/aes_cipher_top.png
[INFO FLW-0001] Saving "placement" to reports/images/aes_cipher_top.placement.png
[INFO FLW-0001] Saving "placement density heatmap" to reports/images/heatmap/placement_density.png
[INFO FLW-0001] Saving "power density for typical heatmap" to reports/images/heatmap/power_density/typical.png
[INFO FLW-0001] Saving "clocks" to reports/images/aes_cipher_top.clocks.png
[INFO FLW-0001] Saving clk clock tree for typical in reports/images/clocktree/clk.typical.png
[INFO FLW-0001] Saving "optimizer" to reports/images/aes_cipher_top.optimizer.png
```

gadfort, 2023-08-26:
The file to add or remove the -skip_gate_cloning to is located in:
`build/aes/job0_freepdk45_demo_orfs/sc_collected_files/scripts_fccbfa790db5ddcc0abfcccac3b21bc686d7535a/sc_cts.tcl`

rovinski, 2023-09-07:
@gadfort resolved?

closed 2023-09-14 by maliberty