The-OpenROAD-Project/OpenROAD #3909: Update io placement to have an option to output the placement to a .tcl file

oharboe, 2023-08-27:
### Description

I'm using routing by abutment for two different macros that need to have the same pin-layout.

If I use `PLACE_PINS_ARGS=-annealing`, I get different pin layout for these two macros.

### Suggested Solution

Add option to io placement to write out the pin-placement to a .tcl file. This .tcl file can then be fed into the second macro and pin layout will be identical.

eder-matheus, 2023-08-28:
@oharboe If it is guaranteed that the two different macros have the same track positions, we can use the `place_pin` command for each IO pin of the design. So the output .tcl file would be a list of calls for the `place_pin` command, like this:

```
place_pin -pin_name pin0 -layer layer -location "x0 y0"
place_pin -pin_name pin1 -layer layer -location "x1 y1"
...
place_pin -pin_name pinN -layer layer -location "xN yN"
```

We could easily group the pins by edge in this format. The only problem is that this file can be huge with one line per pin, but I think this is not a blocker.

About the output for long runtime, do you think an `-timeout` option could help? The annealing run would end after reaching the timeout, and the current result would be the final assignment, followed by a warning message and the suggestion of a new configuration for the constraints.

oharboe, 2023-08-28:
> @oharboe If it is guaranteed that the two different macros have the same track positions, we can use the `place_pin` command for each IO pin of the design. So the output .tcl file would be a list of [...]

:+1: 

This would allow me to cut and paste some pins for my use-case and also to use the annealing.

> The only problem is that this file can be huge with one line per pin, but I think this is not a blocker.

Size doesn't really matter here, no. It's thousands of lines for thousands of pins. That's easily understood.

> 
> About the output for long runtime, do you think an `-timeout` option could help? The annealing run would end after reaching the timeout, and the current result would be the final assignment, followed by a warning message and the suggestion of a new configuration for the constraints.

I think perhaps an "effort" similar to detailed routing/global routing would make sense. I'm thinking this default should be something reasonable, like minutes and then the user can bump it up. The results should be idempotent, so it shouldn't matter how fast your CPU is.

eder-matheus, 2023-08-28:
> > @oharboe If it is guaranteed that the two different macros have the same track positions, we can use the `place_pin` command for each IO pin of the design. So the output .tcl file would be a list of calls for the `place_pin` command, like this:
> > ```
> > place_pin -pin_name pin0 -layer layer -location "x0 y0"
> > place_pin -pin_name pin1 -layer layer -location "x1 y1"
> > ...
> > place_pin -pin_name pinN -layer layer -location "xN yN"
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > We could easily group the pins by edge in this format.
> 
> 👍
> 
> This would allow me to cut and paste some pins for my use-case and also to use the annealing.
> 
> > The only problem is that this file can be huge with one line per pin, but I think this is not a blocker.
> 
> Size doesn't really matter here, no. It's thousands of lines for thousands of pins. That's easily understood.
> 
> > About the output for long runtime, do you think an `-timeout` option could help? The annealing run would end after reaching the timeout, and the current result would be the final assignment, followed by a warning message and the suggestion of a new configuration for the constraints.
> 
> I think perhaps an "effort" similar to detailed routing/global routing would make sense. I'm thinking this default should be something reasonable, like minutes and then the user can bump it up. The results should be idempotent, so it shouldn't matter how fast your CPU is.

Sounds good. I will address this issue in two PRs, starting with the .tcl file with the IO placement and then another PR for the long runtimes.

closed 2023-08-29 by maliberty