The-OpenROAD-Project/OpenROAD #3910: Very large slew for megaboom

oharboe, 2023-08-27:
### Describe the bug

Using https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1398

Run `make synth`, then `make open_synth` and run:

```
report_checks -path_delay max
```

### Expected Behavior

Skew in 10s of picoseconds

### Environment

```shell
latest master
```

### To Reproduce

See above

gadfort, 2023-08-27:
@oharboe how does this compare with the numbers out of place? This is basically just the synthesis numbers which are not particularly great before `repair_design` has been called. Also I see several instances of a 0p5 gate which should not be used in synthesis

maliberty, 2023-08-28:
In floorplan we call remove_buffers.  If you want to look at timing this early you should do so either right after synthesis or after repair_design.

oharboe, 2023-08-28:
From `make gui_place`, the largest slew is now 206.87.

```
>>> report_checks -path_delay max -fields {slew cap}
Startpoint: reset (input port clocked by clock)
Endpoint: AsyncQueueSink/widx_gray_sync_0/reg_0/_09_
          (recovery check against rising-edge clock clock)
Path Group: asynchronous
Path Type: max

    Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------
           0.00    0.00    0.00   clock clock (rise edge)
                   0.00    0.00   clock network delay (ideal)
                1500.00 1500.00 v input external delay
   1.83    0.00    0.00 1500.00 v reset (in)
  16.42   13.47   14.51 1514.51 v input417/Y (BUFx10_ASAP7_75t_R)
  53.01   20.98   39.93 1554.44 v wire138167/Y (BUFx16f_ASAP7_75t_R)
  57.82   12.78   66.29 1620.74 v load_slew138166/Y (BUFx16f_ASAP7_75t_R)
  64.88   35.84   85.90 1706.63 v load_slew138159/Y (BUFx16f_ASAP7_75t_R)
  65.15   35.59   51.41 1758.04 v load_slew138156/Y (BUFx16f_ASAP7_75t_R)
  37.26   18.42   50.36 1808.40 v max_length138155/Y (BUFx16f_ASAP7_75t_R)
  64.30   24.71   54.89 1863.28 v wire138154/Y (BUFx16f_ASAP7_75t_R)
  48.37    8.80   23.45 1886.73 v load_slew138153/Y (BUFx16f_ASAP7_75t_R)
  46.98   15.00   95.80 1982.53 v wire138152/Y (BUFx16f_ASAP7_75t_R)
  48.03   12.36   54.93 2037.46 v wire138151/Y (BUFx16f_ASAP7_75t_R)
  27.12   14.36  105.48 2142.94 v wire138149/Y (BUFx16f_ASAP7_75t_R)
  20.46   11.89   63.47 2206.41 v max_length138148/Y (BUFx12f_ASAP7_75t_R)
   1.23   10.61    8.63 2215.04 ^ AsyncQueueSink/widx_gray_sync_0/reg_0/_05_/Y (INVx1_ASAP7_75t_R)
          10.61    0.02 2215.05 ^ AsyncQueueSink/widx_gray_sync_0/reg_0/_09_/SETN (DFFASRHQNx1_ASAP7_75t_R)
                        2215.05   data arrival time

           0.00 7500.00 7500.00   clock clock (rise edge)
                   0.00 7500.00   clock network delay (ideal)
                   0.00 7500.00   clock reconvergence pessimism
                        7500.00 ^ AsyncQueueSink/widx_gray_sync_0/reg_0/_09_/CLK (DFFASRHQNx1_ASAP7_75t_R)
                  -2.51 7497.49   library recovery time
                        7497.49   data required time
-----------------------------------------------------------------------
                        7497.49   data required time
                        -2215.05   data arrival time
-----------------------------------------------------------------------
                        5282.44   slack (MET)

Startpoint: coreplex/RocketTile/core/rob/_679912_
            (falling edge-triggered flip-flop clocked by clock')
Endpoint: coreplex/RocketTile/core/register_read/_17028_
          (falling edge-triggered flip-flop clocked by clock')
Path Group: clock
Path Type: max

    Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------
           0.00    0.00    0.00   clock clock' (fall edge)
                   0.00    0.00   clock network delay (ideal)
           0.00    0.00    0.00 v coreplex/RocketTile/core/rob/_679912_/CLK (DFFLQNx2_ASAP7_75t_R)
   4.08   24.57   44.89   44.89 ^ coreplex/RocketTile/core/rob/_679912_/QN (DFFLQNx2_ASAP7_75t_R)
   2.74    8.85    7.45   52.35 v coreplex/RocketTile/core/rob/_341445_/Y (INVx4_ASAP7_75t_R)
  31.92    8.82   16.48   68.83 v wire128223/Y (BUFx16f_ASAP7_75t_R)
  36.06   11.14   46.83  115.66 v wire128222/Y (BUFx16f_ASAP7_75t_R)
  20.45   15.77  111.67  227.33 v max_length128221/Y (BUFx12f_ASAP7_75t_R)
  54.13   14.56   51.27  278.59 v wire128220/Y (BUFx16f_ASAP7_75t_R)
  56.41   15.71   93.68  372.28 v load_slew128219/Y (BUFx16f_ASAP7_75t_R)
  56.26   16.77  112.54  484.81 v load_slew128218/Y (BUFx16f_ASAP7_75t_R)
  19.01  206.87  163.97  648.78 ^ coreplex/RocketTile/core/issue_unit/IssueSlot_2/_1830_/Y (AOI211x1_ASAP7_75t_R)
  20.53   72.93  100.20  748.98 ^ coreplex/RocketTile/core/issue_unit/_211429_/Y (AO222x2_ASAP7_75t_R)
  27.87   13.01   30.75  779.73 ^ max_length100249/Y (BUFx16f_ASAP7_75t_R)
  24.09   94.53   53.95  833.68 ^ coreplex/RocketTile/core/issue_unit/_211467_/Y (OR4x2_ASAP7_75t_R)
  21.26   82.62   78.15  911.83 ^ coreplex/RocketTile/core/issue_unit/_211500_/Y (OR5x2_ASAP7_75t_R)
   9.37   48.72   48.10  959.93 v coreplex/RocketTile/core/issue_unit/_211507_/Y (NOR2x2_ASAP7_75t_R)
[... 74 lines cut ...]
  31.17   12.22   68.72 4940.24 ^ wire19594/Y (BUFx16f_ASAP7_75t_R)
   0.71   33.28   72.42 5012.67 v coreplex/RocketTile/core/regfile/_342068_/Y (AOI22x1_ASAP7_75t_R)
   0.97    9.54   21.96 5034.63 v coreplex/RocketTile/core/regfile/_342069_/Y (AND3x1_ASAP7_75t_R)
  18.10   56.13   45.06 5079.69 v coreplex/RocketTile/core/regfile/_342081_/Y (OA33x2_ASAP7_75t_R)
  19.07   73.55  111.25 5190.94 v coreplex/RocketTile/core/regfile/_342126_/Y (OR5x2_ASAP7_75t_R)
   6.86   23.60   63.22 5254.16 v coreplex/RocketTile/core/regfile/_342217_/Y (AO31x2_ASAP7_75t_R)
  20.37    9.97   23.32 5277.48 v wire4635/Y (BUFx12f_ASAP7_75t_R)
   1.66   11.78   59.50 5336.99 v coreplex/RocketTile/core/register_read/_12036_/Y (OA22x2_ASAP7_75t_R)
   0.69    7.18   15.49 5352.48 v coreplex/RocketTile/core/register_read/_12039_/Y (AO21x1_ASAP7_75t_R)
           7.18    0.01 5352.49 v coreplex/RocketTile/core/register_read/_17028_/D (DFFLQNx2_ASAP7_75t_R)
                        5352.49   data arrival time

           0.00 7500.00 7500.00   clock clock' (fall edge)
                   0.00 7500.00   clock network delay (ideal)
                   0.00 7500.00   clock reconvergence pessimism
[... cut ...]           7500.00 v coreplex/RocketTile/core/register_read/_17028_/CLK (DFFLQNx2_ASAP7_75t_R)
                  -9.79 7490.21   library setup time
                        7490.21   data required time
-----------------------------------------------------------------------
                        7490.21   data required time
                        -5352.49   data arrival time
-----------------------------------------------------------------------
                        2137.73   slack (MET)

```

oharboe, 2023-08-28:
There is something very wrong with the megaboom RTL as it stands. There's no way the RTL, as it stands, can get to 1GHz with dozens of levels of logic in Yosys without some heavy rewriting on Yosys' part...

In particular, retiming is mentioned:  https://docs.boom-core.org/en/latest/sections/physical-realization.html#register-retiming

It is possible that the above path could be fixed by breaking it into several pipeline stages with retiming, but I don't have details to know...

Here is an unanswered post that pointed me to some issues...

https://groups.google.com/g/riscv-boom/c/ief9H6WA6sQ

oharboe, 2023-08-28:
abk@ucsd.edu wrote about megaboom and retiming... https://www.tauworkshop.com/2019/slides/TAU2019-Kahng-keynote-final-posted.pdf

closed 2023-08-28 by oharboe