The-OpenROAD-Project/OpenROAD #3933: [WARNING PPL-0036] Number of sections is 1404 while the maximum recommended value is 600 this may negatively affect performance.

oharboe, 2023-09-01:
### Description

Improve error message. What is a "section" and how does it relate to floorplan area?

### Suggested Solution

The error message should give some actionable advice, like "increase area such that width + height is n% greater" or somesuch.

maliberty, 2023-09-01:
Is this with the old ppl?

oharboe, 2023-09-01:
> Is this with the old ppl?

Yes. The error is correct, the area is too small(not enough space on the edges).

This feature request is about.making the error more immediately actionable.

eder-matheus, 2023-09-01:
@oharboe Could you share the complete PPL log? I want to understand the flow of messages you're having in this design.

oharboe, 2023-09-03:
I will reopen if I have the complete logs.

closed 2023-09-03 by oharboe