The-OpenROAD-Project/OpenROAD #3936: tapcell.tcl runs 'forever', [INFO ODB-0303] The initial 119565 rows (-1535139212 sites) were cut with

oharboe, 2023-09-02:
### Describe the bug

```
OpenROAD v2.0-10090-g067d1f4be 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO-FLOW] Tap and End Cap cell insertion
[INFO-FLOW]   TAP Cell          : TAPCELL_ASAP7_75t_R
[INFO-FLOW]   ENDCAP Cell       : TAPCELL_ASAP7_75t_R
[INFO-FLOW]   Halo Around Macro : 2 2
[INFO-FLOW]   TAP Cell Distance : 25
[INFO ODB-0303] The initial 119565 rows (-1535139212 sites) were cut with 18 shapes for a total of 153977 rows (-1894716766 sites).
[INFO TAP-0004] Inserted 239130 endcaps.
```

### Expected Behavior

complete in a reasonable amount of time

### Environment

```shell
OpenROAD v2.0-10090-g067d1f4be
```

### To Reproduce

See above.

vijayank88, 2023-09-02:
@oharboe 
The core area seems very high.
```
>>> ord::get_core_area
2.052 2.16 32284.764 32284.71
```
Is that created by automatically or manual size passed?

I have seen such cases, if users give big die size more than 10k micron, tapcell insertion takes too long.

oharboe, 2023-09-03:
> `CORE_UTILIZATION=0.5` this will increase die size. Normally `30-60` reasonable for automatic die size calculation.

Actually, the design I'm working on has pretty much only macros at the top level. CORE_UTILIZATION misbehaves in this case, giving very, very large die area, even for 0.99 :-)

> I have seen such cases, if users give big die size more than 10k micron, tapcell insertion takes too long.

This github issue is a bug report(or perhaps feature request rather) to fix error messages. There is some sort of overflow, see "(-1535139212 sites)" in the error message.

maliberty, 2023-09-03:
Would you make an example test case for the ifp issue?

maliberty, 2023-09-03:
Note that
```
| `-utilization` | Percentage utilization. Allowed values are `double` in the range `(0-100]`.
```
0.99 is still very low (0.99% not 99%).

oharboe, 2023-09-03:
> Note that
> 
> ```
> | `-utilization` | Percentage utilization. Allowed values are `double` in the range `(0-100]`.
> ```
> 
> 0.99 is still very low (0.99% not 99%).

doh! I saw it next to place density :-)

oharboe, 2023-09-03:
> Would you make an example test case for the ifp issue?

I was messing around with megaboom when I ran into it, bu

oharboe, 2023-09-03:
> Would you make an example test case for the ifp issue?

ifp?

maliberty, 2023-09-03:
Ifp is initialize floorplan. However with your misunderstanding I don't think it matters anymore

eder-matheus, 2023-09-04:
Is this still an issue? The problem with tapcell here is that it needs to place cells in more than 150k rows, each row with almost 600k sites. This seems like a lot, but I don't know if we can have a real design with such large dimensions.

oharboe, 2023-09-04:
> Is this still an issue? The problem with tapcell here is that it needs to place cells in more than 150k rows, each row with almost 600k sites. This seems like a lot, but I don't know if we can have a real design with such large dimensions.

The design is not real, it is a mistake in floorplanning w.r.t. units.

If it isnt worth fixing the overflow in the error message, then I would just close the issue.

Perhaps a clearer error message for such floorplanning mistakes could be worthwhile.

gadfort, 2023-09-04:
@eder-matheus on `initialize_floorplan` it probably would have been good to generate an error when the computed area overflowed (it looks like it might have since the width is ~2^15)

eder-matheus, 2023-09-04:
> @eder-matheus on `initialize_floorplan` it probably would have been good to generate an error when the computed area overflowed (it looks like it might have since the width is ~2^15)

It is not overflowing since we represent the die area with std::int64_t. The value is huge, tho: 1042437131372025.
I will add an error message in the cutRows code to detect the overflow when calculating the total number of sites.

maliberty, 2023-09-05:
This feels too targeted as a fix.  I'm sure there are many places that will fail on such an unreasonably large design.  Also the message doesn't offer much insight into the problem. 

I suggest in initialize_floorplan we error on any block with more than 2^32 sites.

closed 2023-09-12 by maliberty