The-OpenROAD-Project/OpenROAD #3945: Crash on exit

oharboe, 2023-09-05:
### Describe the bug

unzip and run, this opens the GUI with the design, close the OpenROAD GUI window.
[gui_crash.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12523114/gui_crash.zip)

```
$ ./run-me-gcd-nangate45-base.sh 
OpenROAD v2.0-10101-g6e6cd3fad 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Loading spef
[WARNING GUI-0066] Heat map "IR Drop" has not been populated with data.
pure virtual method called
terminate called without an active exception
Signal 6 received
Stack trace:
[ERROR GUI-0077] QThread: Destroyed while thread is still running
terminate called recursively
Signal 6 received
Stack trace:
 0# 0x000055D1EB82801A in openroad
 1# 0x00007F1D9103C4B0 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F1D914A4EE6 in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x00007F1D914B6E9C in /lib/x86_64-linux-gnu/libstdc++.so.6
 7# 0x00007F1D914B6F07 in /lib/x86_64-linux-gnu/libstdc++.so.6
 8# 0x00007F1D914B7C35 in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# gui::RenderThread::getColor(odb::dbTechLayer*) in openroad
10# gui::RenderThread::drawInstanceShapes(odb::dbTechLayer*, QPainter*, std::vector<odb::dbInst*, std::allocator<odb::dbInst*> > const&, odb::Rect const&, gui::GuiPainter&) in openroad
11# gui::RenderThread::drawLayer(QPainter*, odb::dbBlock*, odb::dbTechLayer*, std::vector<odb::dbInst*, std::allocator<odb::dbInst*> > const&, odb::Rect const&, gui::GuiPainter&) in openroad
12# gui::RenderThread::drawBlock(QPainter*, odb::dbBlock*, odb::Rect const&, int) in openroad
13# gui::RenderThread::draw(QImage&, QRect const&, std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> > const&, std::array<std::set<gui::Selected, std::less<gui::Selected>, std::allocator<gui::Selected> >, 8ul> const&, std::vector<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> >, std::allocator<std::unique_ptr<gui::Ruler, std::default_delete<gui::Ruler> > > > const&, double, QColor const&) in openroad
14# gui::RenderThread::run() in openroad
15# 0x00007F1D920CCCC7 in /lib/x86_64-linux-gnu/libQt5Core.so.5
16# 0x00007F1D9108F18A in /lib/x86_64-linux-gnu/libc.so.6
17# 0x00007F1D9111DBD0 in /lib/x86_64-linux-gnu/libc.so.6
./run-me-gcd-nangate45-base.sh: line 8: 200456 Aborted                 (core dumped) openroad -gui -no_init ${SCRIPTS_DIR}/gui.tcl
```

[image]

### Expected Behavior

no crash

### Environment

```shell
OpenROAD v2.0-10101-g6e6cd3fad
```

### To Reproduce

See above

oharboe, 2023-09-05:
@gadfort @maliberty Bisecting:

```
$ git bisect good
18ddd0fa41bdd53e653a592d56e7602903471956 is the first bad commit
commit 18ddd0fa41bdd53e653a592d56e7602903471956
Author: Peter Gadfort <gadfort@zeroasic.com>
Date:   Mon Sep 4 16:19:58 2023 -0400

    gui: restore gui hide behavior
    
    Signed-off-by: Peter Gadfort <gadfort@zeroasic.com>

 src/gui/src/gui.cpp        | 8 ++++----
 src/gui/src/mainWindow.cpp | 1 +
 2 files changed, 5 insertions(+), 4 deletions(-)
```

gadfort, 2023-09-05:
@maliberty I would suggest reverting the change I made. I still think it's "more" correct, but this seems to have broken our CI and recreated the crashing issue. It will take some time for me to dig through the correct behavior we need.

oharboe, 2023-09-05:
> @oharboe is that local or docker based installation?

I only tried local install on ubuntu 23. It looks like @gadfort knows what is going on here.

vijayank88, 2023-09-05:
> > @oharboe is that local or docker based installation?
> 
> I only tried local install on ubuntu 23. It looks like @gadfort knows what is going on here.

Plz mention the OS name part of environment, so it will be understood. Thanks

closed 2023-09-06 by oharboe