The-OpenROAD-Project/OpenROAD #3950: Partition gcd test failing in docker/pre-built binary setup

luarss, 2023-09-06:
### Describe the bug

When I am running the command `partition_gcd.tcl`, it errors out with `Killed` error.

Output matches perfectly of this ground truth file until L4650 of this [log file](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/par/test/partition_gcd.ok). 

Does anyone know what are the memory requirements to run this test?

### Expected Behavior

Should complete without errors. 

### Environment

```shell
Using OpenROAD binary build from Precision Innovations 2023-09-05 in Ubuntu 22.04. OpenROAD using 

OpenROAD binary commit `6e6cd3fadce83be5c8ec345799cce89de20c498d` 

OpenROAD commit `3ab16ae0755a94277a0f79074d4d5ca755b33595`
```

### To Reproduce

- Start docker image `docker run -it --platform linux/amd64 ubuntu:22.04`
- Install OpenROAD binary from https://github.com/Precision-Innovations/OpenROAD/releases
- Clone OpenROAD and cd into par. 
- Run par test: `openroad partition_gcd.tcl`

### Screenshots

[image]

vijayank88, 2023-09-06:
@luarss 
Check your physical RAM usage. Use `htop` in new tab and  `openroad > source partition_gcd.tcl` in another tab, check the memory usage.

Not facing any issue with Ubuntu20.04 and 16gb RAM
```
[INFO PAR-0143] Total number of timing paths = 52
[INFO PAR-0144] Total number of timing-critical paths = 1
[INFO PAR-0145] Total number of timing-noncritical paths = 51
[INFO PAR-0146] The worst number of cuts on timing-critical paths = 1
[INFO PAR-0147] The average number of cuts on timing-critical paths = 1.0
[INFO PAR-0148] Total number of timing-noncritical to timing critical paths = 32
[INFO PAR-0149] The worst number of cuts on timing-non2critical paths = 2
[INFO PAR-0150] The average number of cuts on timing-non2critical paths = 1.28125
===============================================
Exiting TritonPart
Writing partition to verilog.
No differences found.
No differences found.
openroad> 
```

luarss, 2023-09-06:
Yes, confirmed it is a RAM issue. Do you think we should update the minimum specifications in docs?

vijayank88, 2023-09-06:
If user has 8GB RAM with windows and using WSL with docker based means, only 4GB RAM shared with WSL/VM.
So mostly memory issue will be there.

closed 2023-09-06 by luarss

vvbandeira, 2023-09-06:
> If user has 8GB RAM with windows and using WSL with docker based means, only 4GB RAM shared with WSL/VM. So mostly memory issue will be there.

@luarss, this is an important thing to warn the users when using docker. They must ensure that the docker container/VM has access to the minimum required CPU/RAM.