The-OpenROAD-Project/OpenROAD #3964: Detailled routing runs forever when I/O pins on top layer

lustefan97, 2023-09-07:
### Describe the bug

I try to implement a Face-to-face 3D flow. I use the Ibex core design as a base to develop it, I place all its IO pins on the top layer (Pad in ASAP7) but the detailled routing doesn't seem to converge towards an optimized solution with 0 violations. It always "oscillates" around a certain number of violations but never seems to tend to zero.

### Expected Behavior

Detailled routing finishing and optimization converging towards 0 violations.

### Environment

```shell
Git commit: 972137093aa4850b04b51185956b51b8a28b2ccf
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
```

### To Reproduce

[Test3D.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12552899/Test3D.zip)

In this zip file should be all the files needed to reproduce the issue. 
Create a directory "Test3D" in _/OpenROAD/test/_, enter it, open the GUI and enter _source "ibex3D_asap7_bottom.tcl"_ in the GUI TCL command box.

eder-matheus, 2023-09-12:
@lustefan97 Thanks for the test case. Debugging it, I've noticed that some IO pins placed on the PAD layer are very close to each other in adjacent tracks:
[image]

I'm trying a different config for the define_pin_shape_pattern command. Can you try a larger x/y pitch?
I'm running this config now:
```
set top_layer Pad
set top_pin_size {0.04 0.04}
set x_pitch 0.16
set y_pitch 0.16
```

With one track of difference between each pin, I expect the detailed router will have more room to route those pins.

eder-matheus, 2023-09-14:
@lustefan97 I found a good setup for your test case, where the detailed route finishes in 4 iterations with zero DRCs.

I suggest you using this configuration for global and detailed routing:

```
# Global Routing
set global_routing_layers M2-M9
set global_routing_clock_layers M4-M6
set global_routing_layer_adjustments {{M1 0.5} {M2 0.5} {M3 0.5} {M4 0.5} {M5 0.5} {M6 0.5} {M7 0.5} {M8 0.5} {M9 0.5}}

# Detailled Routing
set min_routing_layer M2
set max_routing_layer M9
```

By excluding the layer `Pad` from the range, the router will connect to the pins from below without any wires being created in this layer. Without wires in this layer, fewer DRCs are created during the iterations. If I remember correctly, this layer should not have signal routing.

Also, I've increased the `x_pitch` and `y_pitch` to 0.32 because smaller values will always cause spacing violations:

```
set top_layer Pad
set top_pin_size {0.04 0.04}
set x_pitch 0.32
set y_pitch 0.32
```

I suggest you look at the pin size to ensure the minimum area of the pin is correct. I think it needs to be larger, but I'm not 100% sure.

lustefan97, 2023-09-15:
@eder-matheus Sorry for the late response, I was testing larger pitch between pins (with the old setup). It indeed resulted in lower DRCs but still never converged. 

I have a few questions (some due to being a beginner, I work on this for my master thesis) : 

1. Why do we route only from M2 to M9 and not take into account M1 and Pad layers ? How will the tool handle M1 then ?
2. I tried the setup that you gave me (without updating OpenROAD yet) and it crashed due to some stacked vias. Do those stacked vias are in a configuration going, for example, "M2-V3-M3-V4-M4-V5-etc" ? Is this what has been fixed, to make the tool capable of making a long vertical via towards the IO pins on the TOP layer (by stacking vias and metal layer on the same (x,y) position) even if  they are multiple layers above the maximum ROUTING layer (so say the maximum routing layer is M7, the tool will stack vias and metal layer to connect IOs on the Pad layer ? Was the tool not capable of handling that before ?  Just fyi, here is the log I had (again keep in mind, OpenROAD was not yet updated) : 
``` 
Signal 11 received
Stack trace:
 0# 0x0000565128B2AF7B in openroad
 1# 0x00007F087162E520 in /lib/x86_64-linux-gnu/libc.so.6
 2# triton_route::TritonRoute::stackVias(odb::dbBTerm*, int, int) in openroad
 3# triton_route::TritonRoute::processBTermsAboveTopLayer() in openroad
 4# triton_route::TritonRoute::initDesign() in openroad
 5# triton_route::TritonRoute::pinAccess(std::vector<odb::dbInst*, std::allocator<odb::dbInst*> >) in openroad
 6# pin_access_cmd(char const*, char const*, char const*, int, int) in openroad
 7# 0x000056512934FFAD in openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007F08754C6028 in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# gui::TclCmdInputWidget::executeCommand(QString const&, bool, bool) in openroad
13# gui::CmdInputWidget::handleEnterKeyPress(QKeyEvent*) in openroad
14# gui::TclCmdInputWidget::keyPressEvent(QKeyEvent*) in openroad
15# QWidget::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
16# QFrame::event(QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
17# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# QApplication::notify(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
19# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
20# 0x00007F0872F2902F in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
21# QApplicationPrivate::notify_helper(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
22# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib/x86_64-linux-gnu/libQt5Core.so.5
23# QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
24# QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Gui.so.5
25# 0x00007F086DDE6D6E in /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
26# g_main_context_dispatch in /lib/x86_64-linux-gnu/libglib-2.0.so.0
27# 0x00007F0870CCA258 in /lib/x86_64-linux-gnu/libglib-2.0.so.0
28# g_main_context_iteration in /lib/x86_64-linux-gnu/libglib-2.0.so.0
29# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
30# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib/x86_64-linux-gnu/libQt5Core.so.5
31# QCoreApplication::exec() in /lib/x86_64-linux-gnu/libQt5Core.so.5
32# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in openroad
33# ord::tclAppInit(Tcl_Interp*) in openroad
34# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
35# main in openroad
36# 0x00007F0871615D90 in /lib/x86_64-linux-gnu/libc.so.6
37# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
38# _start in openroad
Segmentation fault
```
3. I indeed, per default, choose to make pins have the minimum size, I will definitely try larger sizes in the future, and I will see what happens

I will update OpenROAD and try again. I will do the same for the top die after that, if I have any issues I will come back to you ! Thank you !

eder-matheus, 2023-09-15:
@lustefan97 I will answer your questions below:

1. The first routing layer usually has many shapes from the placed instances, making it hard to create wires. By setting the first usable routing layer as M2, we ensure that wires will not be created in this layer and access to the M1 pins will be made only through vias. Similarly, we don't want wires in the Pad layers, and the access will be made only through vias.

2. I'm working to fix the crash. I have an open PR to fix it, but I'm also looking at a last issue related to your test case. I hope to have it on master in the next couple of hours. But the crash is related to not having a default via defined for the layer V0. The stack vias process looks for the default vias in every cut layer, but we needed a check for layers that don't have it, leading to this crash. My PR also fixes other bugs besides the crash.
And yes, the tool will stack vias from the max routing layer defined to the pin layer. So if you define the max routing layer as M7, it will have a stack of vias M7-M8-M9-Pad, and the routing will connect to the M7 shape of the via.

I'll let you know when the fix is merged.

eder-matheus, 2023-09-18:
@lustefan97 Could you try your test case again with the latest master branch? My fixes were just merged.

lustefan97, 2023-09-19:
@eder-matheus I just tried my test case and it worked nicely ! 

I do though have another problem seemingly related to the V0 default vias... This test case was for the bottom die and now I tried to do run a flow for the top die (which is basically a copy/paste of the bottom with a few differences : I manually place the IO pins based on the mirrored positions from the bottom that I recover using a file) and I run into pin access issues...

Do I open another issue for this ? I have the exact same setup, I just have a few more/modified lines of code related to the pin placement (open/read/close file, store them into a variable and place them)

lustefan97, 2023-09-19:
Perfect I'll do that !

closed 2023-09-19 by lustefan97