The-OpenROAD-Project/OpenROAD #3982: Unable to connect macro/pad

kareefardi, 2023-09-13:
### Describe the bug

There are two caravel user project designs with multiple power domains. Both have a single macro inside. Also in both cases this macro is connected to a single power domain. However only one design is failing during `analyze_power_grid` with the following error `Unable to connect macro/pad Instance mprj to the power grid`.

### Expected Behavior

I suppose either both should work or both fail. It would be helpful to understand why one design fails and the other work.

### OpenROAD Environment

```shell
Git commit: 5a27687a8c8c5193555bf98395a0a4cdaca3cbd3
kernel: Linux 5.15.0-83-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.1
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10236-g5a27687a8
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
[... 152 lines cut ...]
[INFO]: Creating clock {clk} for port wb_clk_i with period: 25
[INFO]: Setting clock uncertainity to: 0.25
[INFO]: Setting clock transition to: 0.15
[INFO]: Setting maximum transition to: 1.5
[INFO]: Setting maximum fanout to: 10
[INFO]: Setting timing derate to: 5.0 %
[INFO]: Setting clock latency range: 4.65 : 5.57
[INFO]: Setting clock transition: 0.61
[INFO]: Setting RC values...
[INFO PSM-0022] Using 1.800V for vccd1
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[INFO PSM-0031] Number of PDN nodes on net vccd1 = 1260.
[INFO PSM-0064] Number of voltage sources = 4140.
[INFO PSM-0040] All PDN stripes on net vccd1 are connected.
########## IR report #################
Corner: Typical
Worstcase voltage: 1.80e+00 V
Average IR drop  : 3.87e-08 V
Worstcase IR drop: 2.85e-07 V
######################################
[INFO PSM-0022] Using 1.800V for vccd2
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[INFO PSM-0031] Number of PDN nodes on net vccd2 = 806.
[ERROR PSM-0042] Unable to connect macro/pad Instance mprj to the power grid
```

maliberty, 2023-09-13:
@gadfort do you have time to look at this?

gadfort, 2023-09-13:
@maliberty probably not in the next couple of days (or more)

maliberty, 2023-09-13:
I tried the 'not working' case but I get
```
read_liberty -corner Typical pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Error: irdrop.tcl, 15 Traceback (most recent call last):
  File "home/karim/work/OpenLane/scripts/utils/get_default_operating_conditions.py", line 18, in <module>
    from libparse import LibertyParser
ModuleNotFoundError: No module named 'libparse'
```
please fix the test case to be portable.

kareefardi, 2023-09-13:
Re-uploaded both: 
Not working:
[issue_reproducible.partaa.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12602256/issue_reproducible.partaa.tar.gz)
[issue_reproducible.partab.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12602260/issue_reproducible.partab.tar.gz)
Working:
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12602265/issue_reproducible.tar.gz)

maliberty, 2023-09-14:
In your passing test case you have an met4 blockage along with the power pins:
[image]

in not working you don't:
[image]

What is the cause for the difference?  In any case the problem seems to start in pdn and I would need a reproducer for that step to see what's going on.

kareefardi, 2023-09-27:
Here are reproducibles at pdn stage
Design A "Working": 
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12739931/issue_reproducible.tar.gz)
Design B:
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12739939/issue_reproducible.tar.gz)

The reason for the difference is that in design B the macro has no signal routing on met4 so the generated lef has no obstructions over met4.

maliberty, 2023-09-27:
@gadfort can you take a look? If not I'll look next week when I'm back

gadfort, 2023-09-27:
@maliberty I probably dont have time this week. Looking at the pictures I would assume it's the overlapping obs and power pin that isn't handled correctly.

maliberty, 2023-11-28:
@gadfort do you have any time now?  If not I'll look next week as I've got some items that need to be finished by end-Nov.

gadfort, 2023-11-28:
@maliberty I can take a look this afternoon

gadfort, 2023-11-28:
@kareefardi nevermind, I misunderstood the issue. In the not working case, the vccd2 is not connected to anything, I think that is why PSM is complaining, that should be fixable.

maliberty, 2023-11-28:
In pdn or psm?

gadfort, 2023-11-28:
PSM, pdn is connecting the macros correctly

maliberty, 2023-11-28:
@gadfort will you fix that?

maliberty, 2023-11-28:
@gadfort nvm - just saw the PR

PR #4324 (2023-11-28): psm: only error on logically connected macros without connectivity
Closes #3982

Changes: 
- only check for macro nodes when the macro is supposed to be connected to the power grid (adds a test to check for this condition).
closed 2023-11-28 by maliberty