The-OpenROAD-Project/OpenROAD #3985: Add support for Python scripting in GUI

mithro, 2023-09-13:
### Description

Currently the OpenROAD GUI has a "Scripting" window which accepts TCL commands. TCL isn't a commonly known language and not that user friendly, while Python has become one of the most popular languages on the planet and is well praised for its user friendliness.

### Suggested Solution

It would be **_really_** nice if you could select the "Scripting" window to be either the TCL or Python language.

maliberty, 2023-09-13:
There is a start to this in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2762 but I don't think @gadfort has had any time for it recently.

gadfort, 2023-09-13:
@maliberty yeah, that fell through the cracks.
Basically, there are a couple of issues:
- the logging doesn't go through the logger, so that doesn't work right
- exit and hide needs to be reworked to handle the GUI properly