The-OpenROAD-Project/OpenROAD #3999: It would be nice for `save_image` to support a target image width (or height) specified in pixels

mithro, 2023-09-15:
### Description

Currently `save_image` has a `-resolution` argument which is microns per pixel. This means if you want an image a certain number of pixels wide, you have to calculate the value by first extracting the size of your design in microns and then working out a resolution value which gives you the right number of pixels.

It also took me a while to figure out it was microns per pixel and not pixels per micron and thus for 0.13 micron process I'm going to want a value very small number like `0.1` for a high resolution image (and **not** a big number like `10,000`).

### Suggested Solution

It would be nice if you could just specify the image output width in pixels.

`save_image -width 10000px` or `save_image -height 10000px`

AcKoucher, 2023-10-04:
@mithro Wouldn't it be better if we were to specify both dimensions instead of just one?

e. g., `save_image -size 7200x3200`

maliberty, 2023-10-04:
@mithro what happens if the aspect ratio doesn't match the area viewed?  It seems simpler to specify one and let the tool find the other.

PR #4097 (2023-10-14): gui: add support for entering -width in save_image
Resolves #3999

@rovinski FYI, I'm making the correction you pointed out by creating a cache of the max size.
closed 2023-10-14 by maliberty