The-OpenROAD-Project/OpenROAD #4009: Improve PDN warning and error messages

oharboe, 2023-09-18:
### Description

If I add this line:

```
export PDN_TCL               = $(FLOW_HOME)/platforms/asap7/openRoad/pdn/BLOCKS_grid_strategy.tcl
```

These error messages go away:

```
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - foo_2
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - foo_3
[deleted]
[WARNING PDN-0232] CORE_macro_grid_1 - foo_2 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - foo_3 does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
```

### Suggested Solution

Improve the warnings and errors above to give the user more actionable guidance.

maliberty, 2023-09-18:
If you add that line to what?

oharboe, 2023-09-18:
> If you add that line to what?

I add this line to our config.mk file for our design(that I can't share publically), then the error above goes away.

maliberty, 2023-09-18:
Its hard to say what the message should be if I don't know what the problem is.

maliberty, 2023-09-19:
[test.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12665216/test.zip)

@gadfort would you give advise on what the issue and how we could give better messages with the test case attached

gadfort, 2023-09-19:
@maliberty the message is just there to indicate that nothing was generated in a particular grid. No vias, straps, rings, etc. So the powergrid is incomplete. I'll look at the testcase and see if there is anything else it could provide

kareefardi, 2024-06-23:
This still seems to be relevant. Should I file a new issue?

gadfort, 2024-06-23:
@kareefardi if you have a specific issue, then I would suggest filing a new issue. If you have suggestions for how to improve the error message I would be open to suggestions

oharboe, 2024-06-23:
I think it would be easier to understand what is going on if the failed result was saved with e.g. 2_error.odb and then one can look at the failed result visually, it would explain failed macro placement better than a thousand carefully articulated error messages... this is a compliment to good error messages of course ...

kareefardi, 2024-06-24:
@gadfort The current message indicates that the PDN failed to generate but it doesn't say how. In my experience the failure could be for a couple of reasons:

- No overlap between stripes and macros
- Overlap but impossible to insert vias at such location (due to bad dimensions for example and in such case actually there is no error displayed) 
- Overlap but no connect statements provided
- Some other misc issues

In itself, the message is a bit confusing. I am assuming it means that the grid (not the macro) is empty and doesn't have shapes. I am not sure what is the proper action here but at least as @oharboe is mentioning, having a layout with a PDN that failed to establish this connection would help debug the reason for failure.

gadfort, 2025-08-12:
> @gadfort is anything possible here or should we close?

@maliberty without a concrete suggestion for improvements, I think this can be closed

closed 2025-08-12 by maliberty