The-OpenROAD-Project/OpenROAD #4014: set_domain_area floorplan not fit into GUI

vijayank88, 2023-09-19:
### Describe the bug

I've executed the [upf_test.tcl](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/ifp/test/upf_test.tcl) with ifp test case.
The power domain area seems not fit into OpenROAD gui.

[image]

### Expected Behavior

Fit all relevant db into OpenROAD gui.

### Environment

```shell
With latest master
```

### To Reproduce

```
cd src/ifp/test
openroad upf_test.tcl
```

maliberty, 2023-09-19:
I think it is just a poorly constructed test case as having objects outside the die area doesn't make much sense.   Perhaps that should just be an error at the end of ifp.

gadfort, 2023-09-19:
@maliberty what would the error condition be? the power domain is defined before the floorplan.

maliberty, 2023-09-19:
@gadfort the error would be from the end of initialize_floorplan that there are objects (here regions) outside the die area

gadfort, 2023-09-19:
does `initialize_floorplan` create the domain areas? If so, I would say yes (or initialize_floorplan  should scale correctly to get the domains included

maliberty, 2023-09-19:
@gadfort it doesn't create them but they exist beforehand (at least in this test) so we can check them.

rovinski, 2023-09-19:
Seems like it should definitely be an error somewhere.

vijayank88, 2023-09-30:
@maliberty 
Is this resolved?

I am getting blank screen in GUI. Not able to reproduce this now.

maliberty, 2023-10-02:
I fixed the test case.  If I turn on rows I see

[image]

vijayank88, 2023-10-03:
@maliberty 
I've updated to latest master and ran the test case. If I load with `gui::show` it return empty.
Also can you help to restore OpenROAD gui with default background. I'm getting grey color background.
[image]

maliberty, 2023-10-03:
I don't see the gray screen you describe:
[image]

On what OS are you running?  What exact set of steps are you doing?

vijayank88, 2023-10-04:
CentOS 7.. GCP machine..
Vitor have have access to that machine as admin.

maliberty, 2023-10-04:
@vvbandeira please help me to get access to reproduce this issue.  I recall Seyi had something similar a while ago.

vijayank88, 2023-10-14:
I checked with commit ID 46f712bf130726641921fca4b3a6f6940eb1061b and it's got resolved now.

closed 2023-10-14 by vijayank88