The-OpenROAD-Project/OpenROAD #4015: Error encountered at routing stage: No access point

Multimuffin, 2023-09-19:
### Describe the bug

I am currently trying to integrate a new platform (tower) into the flow, and I am mainly referring to [this](https://openroad-flow-scripts.readthedocs.io/en/latest/contrib/PlatformBringUp.html) tutorial. For the  `config.mk` file in the platform directory, I simply copied the adequate file from the `sky130hd` platform and changed it accordingly, as seen below.
I have managed to reach the routing stage in the flow, but I cannot get past the following error message:

```
[INFO DRT-0165] Start pin access.
[ERROR DRT-0073] No access point for input35/A.
[ERROR DRT-0073] No access point for _135_/A.
[ERROR DRT-0073] No access point for _288_/D.
[ERROR DRT-0073] No access point for _287_/D.
Error: detail_route.tcl, 70 DRT-0073
Command exited with non-zero status 1
```

Btw, as the input file, I have used the design example [spm](https://raw.githubusercontent.com/The-OpenROAD-Project/OpenLane/master/designs/spm/src/spm.v) provided by OpenROAD.

Thanks for any help in advance; I would really appreciate it!

### Expected Behavior

A fully functional flow, that gets past the routing stage.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-31-generic
os: Ubuntu 22.04.3 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10111-gb8c71cc9a
-- System name: Linux
-- Compiler: GNU 11.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 292 lines cut ...]
export CDL_FILE = $(PLATFORM_DIR)/cdl/$(PLATFORM).cdl
export KLAYOUT_LVS_FILE = $(PLATFORM_DIR)/lvs/$(PLATFORM).lylvs
```
[config.mk.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/12658402/config.mk.log)

### **stucture of the platform folder**

```
platform
 |-- asap7
 |--  ...
 |-- sky130ram
 +-- tower
     |-- cells_latch.v
     |-- config.mk
     |-- constraint.sdc
     |-- fastroute.sdc
     |-- gds
     |-- lef
     |-- lib
     |-- make_tracks.tcl
     |-- pdn.tcl
     |-- setRC.tcl
     +-- tapcell.tcl
```

vijayank88, 2023-09-19:
@Multimuffin 
Can you please run `make gui_cts` and check respective pins and its access?

Multimuffin, 2023-09-19:
[image]

[image]

[image]

Here are the cells and the pins that are causing the problems

maliberty, 2023-09-19:
Is this a PDK that can be shared publicly?  If so a test case would be helpful.

If not, you can try adding
```
detailed_route_debug -pa -pa_markers -pin _135_:A
```
to scripts/detail_route.tcl before the router is invoked.  If you run with the GUI it will stop on this pin and show you all the attempted pin access and the associated drc errors.  Usually that gives a good clue as to what is going wrong.

Multimuffin, 2023-09-26:
Unfortunately not but I have figured out how to "solve" (or rather avoid) this issue. At first, I realized, that I forgot to adapt the values in the` pdn.tcl` file which caused some minor issues. But in the end, I was just adding the problematic cells to the `DONT_USE_CELLS` and the flow was running without any problems.
In the future I will try to figure out, what the exact reason for this issue has been but right now I am just happy, that I have got the flow up and running.
Nevertheless, thank you for the help, I appreciate that you are spending your time helping others

closed 2023-09-26 by Multimuffin