The-OpenROAD-Project/OpenROAD #4017: Cut layer V0 has no default via defined & No access point for PIN

lustefan97, 2023-09-19:
### Describe the bug

This is a follow up to #3964.

I wanted to try and fully route a design using the mirrored pin position from the last design. In #3964 the goal was to design the bottom die and now I want to do the same for the top die. 

In order to allow the pins positions to match when both dies are connected "in the real world" I mirror their positions with regards to the x axis, which is done during the "bottom die flow". Their positions are written in a file which contains their name, x/y position and width/length. This file is read at the beginning of the flow and the data is stored into a variable _pins_ .
The flow is executed normally as it is for the most part a copy/paste from the bottom die, and all parameters are the same. The only differences is that for the non-random I/O pin placement I place them manually using the _place_pin_ command and a for loop on _pins_ .

During global routing, I first have this warning multiple times : 

`[WARNING DRT-0204] Cut layer V0 has no default via defined.`

And after checking for pin access : 

`[ERROR DRT-0074] No access point for PIN/the name of the IO pin .`

### Expected Behavior

No warning for V0 default via and pass pin access (and routing after that).

### Environment

```shell
Git commit: 765caaf66d5cd9c32814933b022e369e0865f4b8
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
```

### To Reproduce

[Test3D.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12662260/Test3D.zip)

In this zip file should be all the files needed to reproduce the issue.
Create a directory "Test3D" in /OpenROAD/test/, enter it, open the GUI and enter source "ibex3D_asap7_top.tcl" in the GUI TCL command box.

eder-matheus, 2023-09-20:
@lustefan97 I got this error when running your test case: `Error: TopDieTest.tcl, 21 couldn't open "miscellaneous/topDieFile.pin": no such file or directory`.

I've copied this file from your previous test case, but the detailed router ends with no errors with this file. Could you attach the correct `topDieFile.pin`? 

I've also noticed that you set the max routing layer for the detailed router as Pad. Consider using M9 for it to access the Pad pins from below.

lustefan97, 2023-09-20:
@eder-matheus I think I may have only changed the tcl files without updating the rest in the test case I gave you I'm sorry... I have to delete certain files from the directory in order to be able to upload it here, so I took the older test case and forgot to update the rest.

Regarding the pin access issue (not the crash), I looked into it again, could it be due to the way I mirror the pins ? I mirror them using the core area x coordinates from the top right corner of the die, may the pin access issue be due to a misalignment of the IO pins and routing tracks ? And if so, is it possible to access the coordinates of the right-most track to mirror everything with them ?

eder-matheus, 2023-09-20:
@lustefan97 I think it is related to the pin shapes not being aligned in the routing tracks but not necessarily in the PAD layers. When I use the Pad as the max routing layer, pin access finishes without errors. But when using M9 as the max routing layer, the pin access error happens. DRT will try to find pin access points in the M9 via shape that connects to the Pad pin.
However, pin access works again when setting the M8 as the max routing layer. You can try setting the max routing layer as M8 as a quick fix. The router found pin access points for the via shape in M8, which doesn't affect the router regarding DRCs.

eder-matheus, 2023-09-20:
Btw, the crash that you have is related to the GUI, not with the filler cell placement. This is a known issue and we're investigating to solve it. Meanwhile, you can use the GUI after the flow ends with the command `gui::show`.

lustefan97, 2023-09-24:
I see, I will close this issue then, and look on a better way to mirror the pins. Thanks for your help !

closed 2023-09-24 by lustefan97