The-OpenROAD-Project/OpenROAD #4032: when zooming into macro with large offset, they disappear from view

stefanottili, 2023-09-24:
### Describe the bug

This is a follow up of #3998.

When zooming in, the contents of macros with large offset disappears from view.

Same behavior whether top references macros for a, b and c or whether it's hierarchical def.

It seems that the "is in view" calculation doesn't properly take offset into account.

### Expected Behavior

Display should always show the macro contents.

### Environment

```shell
v2.0-10398-gecc03c290 Sept 23
M1 compiled with mpl2/par patch
```

### To Reproduce

make or 

and then zoom into macro b or c and you eventually get a blank screen.

[test_origin_hier.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12707685/test_origin_hier.tgz)

AcKoucher, 2023-09-29:
@stefanottili

In block.def you are setting the ORIGIN pt and the OBS RECT for each of your BLOCKS using as reference the top.def die area ( 0 0 1000 1000 ), but each BLOCK has its own die and its ORIGIN and OBS should be set accordingly. 

As a result:
Even though the blocks are being drawn in their desired position, they are misplaced. So when we zoom, and the viewer can no longer see the real position of the macro, its blockage disappears.

Using your block.def:
[image]

Setting origin and obs according with its own die:
[image]

stefanottili, 2023-09-29:
Having large negative origin is unusual, but perfectly legal.
The blocks are not misplaced.

The lef/def input data is correct and OR displays's them correctly when doing a fit too.

As you notices correctly, the bbox calculation for b and c is wrong.
It doesn't take the ORIGIN into account and thus the bboxes of b and c are not correct.
That's the reason why zooming into them makes the disappear.

AcKoucher, 2023-09-29:
If that's truly the case, apparently this is not an issue related to the gui, but to odb.

stefanottili, 2023-10-14:
it seems that LAYER <layername> TYPE OVERLAP ; RECT/POLYGON  define the "valid" area of the MACRO.
I've attached the fixed testcase, even though OR currently doesn't support non square DIEAREA or MACRO.

[test_origin_hier.tgz](https://github.com/The-OpenROAD-Project/OpenROAD/files/12901750/test_origin_hier.tgz)

PR #4895 (2024-04-03): odb: avoid skipping bbox computation for inst when block is not valid
Fix #4032

The callback cpp test was considering that, for a created inst whose block has not a valid bbox, calling setOrigin would only be effective in the first call. However, with the fix here, as long as the block has not a valid bbox, every setOrigin call should trigger callback events.
closed 2024-04-03 by maliberty