The-OpenROAD-Project/OpenROAD #4033: zoom in/out to the position of the mouse in the view, not the view center

stefanottili, 2023-09-24:
### Description

The current OpenRoad behavior is to zoom in/out of the center of the view.

It's much more useful when the point to zoom in or out to is determined by the position of the mouse cursor in the view.
This is the default behavior of klayout.

maliberty, 2023-09-24:
It has always zoomed around the mouse point. If that's changed it is a new bug

AcKoucher, 2023-09-24:
@stefanottili Zooming with Ctrl+Wheel already zooms in/out based on the cursor position.

So, to be clear, what is being reported here is that zooming in/out with keyboard should have the same behavior?

stefanottili, 2023-09-25:
Yes

maliberty, 2023-09-25:
That seems much less intuitive to me. When I navigate from the keyboard I don't expect the mouse to be involved

stefanottili, 2023-09-25:
klayout's behavior is when you don't have the mouse in the layout view, use the center of the view for z/Z (zoom in/out).
if the mouse is in the layout view, zoom in/out at the mouse location. 
Which I find very intuitive.
klayout also doesn't require the ctrl-key to pressed to use the mouse wheel.
In that sense it doesn't require the keyboard to be involved when using the mouse to zoom.
And with fit and zoom out gesture (move mouse up right/up left) you really don't need the keyboard.

stefanottili, 2023-09-27:
Thanks for working on this, but unfortunately it doesn't quite work as expected.

There seems to be some zoom out limit on the positioning of a chip in the view that get's in the way.

After a "fit" point the mouse cursor to the bottom right corner of the chip.
Hitting z to zoom in the expectation is that the point the cursor points to will stay under the cursor, 
so subsequent z will keep on zooming in on where the cursor points.

This only works once the zoom level reaches a certain point, but it should work on a full chip view too.

AcKoucher, 2023-09-27:
This could be another issue, as this has to do with how the function that handles the zoom itself works. Zooming with mouse should result in the same problem.

stefanottili, 2023-10-03:
How to proceed here ?
This feature request has been closed even though the feature doesn’t quite work for points at the edge of the view. The point the mouse is pointing to doesn’t stay under the cursor.
Am I supposed to create a new bug report ?

maliberty, 2023-10-03:
The problem seems closely related so I'll reopen.  It does occur with the mouse as well.

PR #4043 (2023-09-27): gui: zoom triggered by keyboard now focus on cursor
Resolves #4033
PR #4767 (2024-03-12): gui: resize viewer based on its actual size rather than the padded rect dimensions
Fix #4033
closed 2024-03-12 by maliberty