The-OpenROAD-Project/OpenROAD #4034: Use gestures not only for zoom in, but also for fit and zoom out

stefanottili, 2023-09-24:
### Description

Currently mouse click and move will zoom in, drawing a rectangle on which area to zoom into.

Klayout has move up right for fit and move up left for zoom out.
This is very useful and convenient for navigating with the mouse.
And it would be nice if the two OpenRoad flow gui”s would behave the same.

What klayout is missing is changing the cursor to make this behavior self explanatory.
Nlview uses “fit” and “x2”, “x3” as cursor for the fit and zoom out gesture.

maliberty, 2024-03-12:
You might look at https://doc.qt.io/qt-5/gestures-overview.html for a custom gesture.

stefanottili, 2024-03-12:
OpenRoad already handles the “zoom in” gesture by drawing a rectangle, so implementing a fit if “move right up” and zoom out for “move left up” adoesnt need any more sophisticated gesture recognition.

stefanottili, 2026-09-22:
keep it open