The-OpenROAD-Project/OpenROAD #4035: dependency installer fails in Ubuntu20.04

vijayank88, 2023-09-25:
### Describe the bug

As equivalence check added in OpenROAD, trying to install those with `sudo ./etc/DependencyInstaller.sh`. but it fails to complete the installation.

### Expected Behavior

Complete the Dependency installation.

### Environment

```shell
Git commit: d0bd5cb959f6655f3efbca11572b24a3f5f3e12e
kernel: Linux 5.15.0-84-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.25.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-10415-gd0bd5cb95
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.5 (found version "3.5.0") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
[... 104 lines cut ...]
You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 0cc2ff0 Merge pull request #46 from YosysHQ/jix/dependent-shell-pattern-escape
yosys-config --build src/eqy_combine.so  src/eqy_combine.cc 
yosys-config --build src/eqy_partition.so  src/eqy_partition.cc 
src/eqy_partition.cc:24:10: fatal error: kernel/json.h: No such file or directory
   24 | #include "kernel/json.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:29: src/eqy_partition.so] Error 1
make: *** Waiting for unfinished jobs....
```

gudeh, 2023-09-25:
I am having the same issue here. I am using "sudo ./setup.sh":

```
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
CMake already installed.
Swig already installed.
Boost already installed.
Eigen already installed.
CUSP already installed.
Lemon already installed.
spdlog already installed.
Cloning into 'eqy'...
remote: Enumerating objects: 1720, done.
remote: Counting objects: 100% (183/183), done.
remote: Compressing objects: 100% (138/138), done.
remote: Total 1720 (delta 63), reused 74 (delta 42), pack-reused 1537
Receiving objects: 100% (1720/1720), 1.04 MiB | 4.92 MiB/s, done.
Resolving deltas: 100% (1100/1100), done.
Note: switching to '0cc2ff0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 0cc2ff0 Merge pull request #46 from YosysHQ/jix/dependent-shell-pattern-escape
yosys-config --build src/eqy_combine.so  src/eqy_combine.cc 
yosys-config --build src/eqy_partition.so  src/eqy_partition.cc 
yosys-config --build src/eqy_recode.so  src/eqy_recode.cc 
src/eqy_partition.cc:24:10: fatal error: kernel/json.h: No such file or directory
   24 | #include "kernel/json.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:29: src/eqy_partition.so] Error 1
make: *** Waiting for unfinished jobs....
```

The error message seems to be the same.

vvbandeira, 2023-09-25:
I cannot reproduce the error using the latest version of OR and a clean image from Docker. I will spin up a VM later to see if there's any difference, but it is unlikely. @vijayank88 can you manually install `eqy`?

gadfort, 2023-09-25:
@vvbandeira @vijayank88 would it be possible to make yosys, sbu, eqy compiles and installs optional for runtime use, it seems like a huge overhead to make everyone compile it when its only used in a few CI tests?

vvbandeira, 2023-09-25:
@gadfort, do you think one switch for all three (e.g., `-equivalence`) or would one for each tool be better?

gadfort, 2023-09-25:
@vvbandeira I think one switch would be fine since most folks that would need it will need to install all three.

vijayank88, 2023-09-26:
@vvbandeira 
Currently I'm using this pre-built package of yosys for Centos7/Ubuntu20.04 to avoid installation issue: https://github.com/YosysHQ/oss-cad-suite-build#installation

gadfort, 2023-09-26:
@vvbandeira any updates on this? We are getting similar failures

vvbandeira, 2023-09-26:
@gadfort I have started #4052 to skip `yosys/eqy/sby` installs

troibe, 2023-09-26:
The issue is on a user machine they most likely have yosys maybe installed from a previous install of openroad.
In this case the yosys version might be older than 0.33 which seems to be required for eqy and sby.
The script does not update yosys if it's already installed (a reasonable default).
**Installing yosys 0.33 solved the issue for me.**
If this is only for CI testing then skipping eqy and sby by default sounds like a good idea.

vvbandeira, 2023-09-26:
Thanks for the input @troibe, @vijayank88 could you check if that is the reason why your install is failing?

Currently, we are enabling `eqy` in the CI tests, but the goal is to provide a way for users to run equivalence checks (if the dependencies are available), somewhat like `report_check`.

vijayank88, 2023-09-27:
Thanks @vvbandeira. With latest commit I am not facing any issues while executing `sudo ./etc/DependencyInstaller.sh`.

vijayank88, 2023-09-27:
Is any switch need to be added to install OpenROAD equivalence dependencies yosys/eqy/sby in general during `sudo ./etc/DependencyInstaller.sh` execution?
https://github.com/The-OpenROAD-Project/OpenROAD/blob/347459a5c91f473de611c0c0f403eb4880e8d339/etc/DependencyInstaller.sh#L11

vvbandeira, 2023-09-27:
@vijayank88 
```
./etc/DependencyInstaller.sh -common -eqy
```

gudeh, 2023-09-28:
Hey guys, with the latest changes I was able to run "sudo ./setup.sh" (from ORFS) without any error messages. Although, when I run the flow I am getting the following error:

```
Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0094] Found 43 endpoints with setup violations.
[INFO RSZ-0040] Inserted 22 buffers.
[INFO RSZ-0041] Resized 107 instances.
[INFO RSZ-0043] Swapped pins on 40 instances.
[INFO RSZ-0049] Cloned 1 instances.
[WARNING RSZ-0062] Unable to repair all setup violations.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 110 couldn't execute "eqy": no such file or directory
Command exited with non-zero status 1
Elapsed time: 0:06.42[h:]min:sec. CPU time: user 6.38 sys 0.03 (99%). Peak memory: 110052KB.
make[1]: *** [Makefile:673: do-4_1_cts] Error 1
make: *** [Makefile:673: results/nangate45/gcd/base/4_1_cts.odb] Error 2
```

Should I create a new issue on ORFS repository with more details (for example the build_openroad.log), or would anyone have a possible easy solution?

Furthermore, as mentioned by @troibe, my Yosys version is 0.33 (as it seems it should be).

vvbandeira, 2024-01-16:
Closing as it is currently working on CI and fixes for `eqy` issues have been merged. Please re-open if the problem persists or start a new issue after testing with the latest code.

closed 2024-01-16 by vvbandeira