The-OpenROAD-Project/OpenROAD #4036: Is there a way to place a power pin rectangle using OpenROAD?

mo-hosni, 2023-09-25:
### Description

I am using OpenROAD through OpenLane and the way power pins are placed are using PDNgen which only places rings or stripes in OpenLane. So, is there a way I can place a power pin rectangle in a specific location?

eder-matheus, 2023-09-25:
@mo-hosni I'm not sure if it will fit what you want, but we have the `place_pin` command, which places an individual IO pin in specific locations. You can also define the dimensions of the pin with this command. Here's the command description: https://github.com/The-OpenROAD-Project/OpenROAD/tree/master/src/ppl#place-individual-pin

maliberty, 2023-09-26:
More of a description of your use case would be helpful.

mo-hosni, 2023-10-09:
The use case is:
I am hardening a macro that will be placed inside a padframe. All the pins will be connected by abutment. So, all the signal pins are placed in the correct locations using the usual IO placer except for the power pins as the power pins are inserted as rings or strips using PDNgen. So, I was asking if there is a way to place a rectangle pin for power and PDNgen connect it to the rings of that power domain,

maliberty, 2023-10-09:
There is no such facility as that's an usual use case.  Usually blocks in the padframe are custom designs and not P&R.  What sort of block is it?

You can directly create shapes through database APIs yourself.   Roughly
```
set block [ord::get_db_block]
set net [$block findNet <name>]
set swire [odb::dbSWire_create $net FIXED]
set tech [ord::get_db_tech]
set layer [$tech findLayer <name>]
odb::dbSBox_create $swire $layer <x1> <y1> <x2> <y2> FIXED
```

closed 2023-10-17 by maliberty