The-OpenROAD-Project/OpenROAD #4041: Add library of useful functions for doing manual placement of cells

mithro, 2023-09-25:
### Description

It would be nice if the OpenROAD tool provides a set of functions to the TCL & Python scripting interface which made it quick and easy to do manual placement of a certain sets of cells. 

Some ideas include;

 * `place_column` -- Place the selected cells into a vertical column. Options could include;
     *  `packed` for placing them as close together as possible.
     *  `equal` for placing them equally distributed between top and bottom coordinates.
     * `global` for placing them in the rough locations which need to be legalized into actual valid positions.
     * `legalize` (or maybe `detailed`) for placing them directly into legalized positions.
 * `place_row` -- Place the selected cells into a horizontal row
     *  Same as `place_column`.
 * `place_array` -- Place the selected cells into a grid / array pattern.
     *  Same as `place_column`.

Some related functions could include;

 * `legalize_cells` -- Legalize the set of cells into correct placements.
 * `route_net` -- Route the selected set of nets.

This are likely to be very useful in things like;

 * Automated analog like OpenFASoC
 * Building blocks like DFFRAM does
 * Distributing cells for wave style computation
 * Placing input / output flops in certain locations
 * Other things I haven't thought of

Dolu1990, 2024-01-05:
Hi,

I think I hit the same issue. I have congestions problems on a register-file using sky130 (doesn't pass at 20% density), and wanted to try helping the place and route by providing "fixed location" in column for the register-file latches.

mithro, 2024-01-05:
@Dolu1990 - You might find using [DFFRAM](https://github.com/AUCOHL/DFFRAM) as an alternative solution.

maliberty, 2024-01-05:
Note that DFFRAM is rather skywater specific and some layouts are far from optimal.

Dolu1990, 2024-01-05:
@mithro Thanks ^^
It seems like DFFRAM won't work for my usecase (i would need 4R 2W and more)

mithro, 2024-01-05:
@Dolu1990 - You should know how to build a 4R 2W out of DFFRAM blocks, same thing we do as in FPGAs -- https://tomverbeure.github.io/2019/08/03/Multiport-Memories.html :-)

mithro, 2024-01-05:
@Dolu1990 - You should also log a feature request on the DFFRAM repository. BTW If you are doing this as part of funded research, funding AUC is extremely cheap.

Dolu1990, 2024-01-06:
@mithro 

> You should know how to build a 4R 2W out of DFFRAM blocks, same thing we do as in FPGAs

Yes right ^^

> You should also log a feature request on the DFFRAM repository

Done : https://github.com/AUCOHL/DFFRAM/issues/192

> If you are doing this as part of funded research, funding AUC is extremely cheap.

It is part of a small exploratory project (~100h work) to see where are the bottleneck in porting NaxRiscv to asic, not enough left for it, but thanks for the tips, i keep it in mind :)

maliberty, 2024-01-06:
Fwiw I did start a rewrite of DFFRAM in OR as a spare time project.  If someone has interest in picking it up I would be glad to share the in progress code.

rovinski, 2024-01-06:
@maliberty that might be a good student project? If you share the code I can determine further.

maliberty, 2024-01-07:
fwiw I put the code at https://github.com/The-OpenROAD-Project/OpenROAD/pull/4496  - this is not complete and is not proposed a solution to anything in its current form.  If someone wants to pick it up and complete it that would be great.

QuantamHD, 2024-01-07:
We should take a page from CSS Grid https://css-tricks.com/snippets/css/complete-guide-grid/ if we're looking for more generic layout primitives.

Ishtiyaque-Alam, 2026-03-04:
@maliberty I d be glad to take up this issue , if its still available.

Divinesoumyadip, 2026-03-14:
> Will attach the PR tomorrow

Hey @Ishtiyaque-Alam, totally understand about the exams , good luck with the rest of them!

Just wanted to give you a heads-up that I’ve finished the heavy lifting on PR #9764. 

To avoid us both duplicating the same work, it might be better to keep the core `place_row/place_column` in that PR. However, there’s still plenty of room to expand this utility! If you’re interested, you could build on top of this foundation to implement the `place_array` or` legalize_cells` features mentioned in the original description. That would be a huge addition to the library!

Ishtiyaque-Alam, 2026-03-14:
Has that PR been reviewed?

Divinesoumyadip, 2026-03-14:
> Has that PR been reviewed?

It’s currently awaiting maintainer review, but all 15/15 CI checks are passing, including `tclint` and the Jenkins build.