The-OpenROAD-Project/OpenROAD #4050: OR gui: create a switch to be able to toggle PIN geometry

stefanottili, 2023-09-26:
### Description

Currently a def PIN and NET are both displayed depending on the Net switch in the gui.

Trying to debug https://github.com/KLayout/klayout/issues/1490 I would have liked to be able to switch the pins of and see where the net is.

### Suggested Solution

add pin switch

maliberty, 2023-09-27:
I don't believe this is accurate though I'm not in front of a computer. There is a control for terms

stefanottili, 2023-09-27:
There is Pin Markers, but that toggles the pinname and the triangle showing the direction.
It would be nice if it could also toggle the pin geometry.

AcKoucher, 2023-09-30:
@stefanottili We have currently in display control: Misc -> Instances -> Pins. That won't allow selection, but the geometry is visible.

stefanottili, 2023-09-30:
I was not looking at instance pins, I was looking at the PINS section of a block.

Nets (also switches "Pin geometry" visibility)
Pin Markers (switches "Pin Name" and "Pin Direction")
Misc->Instances->Pins
Misc->Instances->Pin labels

"Pin Markers" switches visibility of "Pin Name" and "Pin Direction"
The "Pin geometry" visibility gets switched by Nets.

I would find it much more logical to have "Pins" instead of "Pin Label" and that it would switch visibility of "Pin Name", "Pin Direction" and "Pin geometry".

All that's needed is to rename "Pin Labels" to "Pins" and move the visibily switch for pin geometry from "Nets" to "Pins".
And if you really want to treat Nets and Pins the same, add the selection switch to Pins too.

maliberty, 2023-10-02:
I'm ok with this.  @gadfort any thoughts?

gadfort, 2023-10-02:
Seems reasonable

closed 2023-10-11 by maliberty

stefanottili, 2023-10-11:
loading mock-array lef/def into klayout shows that the net routes to the edge of the pin.

In openroad, the "Net" switch also draws the pin geometry, so one can't distinguish pin from net.

[image]

openroad -gui mock-array.or

cat mock-array.or:
read_lef platforms/asap7/lef/asap7_tech_1x_201209.lef
read_lef platforms/asap7/lef/asap7sc7p5t_28_R_1x_220121a.lef
read_lef results/asap7/mock-array_Element/base/Element.lef
read_def results/asap7/mock-array/base/6_final.def
read_def -child results/asap7/mock-array_Element/base/6_final.def

maliberty, 2023-10-11:
Turn off the net and you'll see the pin.

PR #4122 (2023-10-11): gui: use one control for drawing bterms' marker, geometry and text in the layout
#Resolves #4050

Also renaming "Misc->Instances->Pin labels" to "Misc->Instances->Pin Name".