The-OpenROAD-Project/OpenROAD #4068: Add GUI moving and saving of macro placement

oharboe, 2023-10-03:
### Description

Add support for moving and rotating macros from the GUI, snapping to the grid as well as saving of macro placement.

### Suggested Solution

1. grab and move macro, by default snapping to the grid so that detailed router and other stages do not run into trouble.
2. add a `write_placement foo.tcl` command.

rovinski, 2023-10-03:
Snap to which grid? A macro can have arbitrary pin sizes and shapes and it would be extremely difficult to ensure that an arbitrary macro is snapped to the routing grid. Also, pins on different layers could want to snap to different tracks so that at least 1 layer isn't aligned.

There's a lot of potential for this feature to cause frustration if the macro doesn't follow standard conventions exactly. If you supply some more details about the expected behavior, then I imagine this request could gain better traction.

oharboe, 2023-10-03:
@rovinski I am afraid I dont know exactly what the constraints are, I was hoping some expert domain knowledge of this could be built into the GUI

rovinski, 2023-10-03:
Well, you know how with Microsoft Word or PowerPoint when you move an image around and it snaps to everything except how you want it? That's what I am worried about. 😅

Here's an example of a macro with no good way to snap because of pins off track:
[image]

Here's an example of no good way to snap because of pins and tracks misaligned between layers
[image]

Here's an example of very difficult to determine snapping because of polygon pin shapes
[image]

Here's an example of too many potential snapping points because of wide pins
[image]

So I fear that checking for all of these arbitrary conditions would be difficult. Realistically there would have to be a lot of checking to see if a macro could be easily snapped to the routing grid.
Snapping to the placement grid or the manufacturing grid would be much more straightforward.

maliberty, 2023-10-03:
Based on the discussion in #4069 I'm dropping the GUI editing and suggesting we just add a write_macro_placement  command to mpl2.

maliberty, 2023-10-07:
I believe the intention would be to save all the macro placements into one file.

PR #4120 (2023-10-12): mpl2: add command for writing macro placement
Resolves #4068 

In this PR I'm also implementing the command `place_macro` to place macros individually.
The output file of write_macro_placement will be written in the format of `place_macro` commands
as we currently do in PPL.
closed 2023-10-12 by maliberty