The-OpenROAD-Project/OpenROAD #4069: Add support in GUI for moving and saving IO placement

oharboe, 2023-10-03:
### Description

Add support in GUI to manually move and to save the IO placement.

Initially, a constraint file might be written, but in later stages, manually moving pins can be quite useful.

### Suggested Solution

1. Add possibly to move pin, snapping to grids such that detailed placement and other stages do not run into trouble.
2. add `write_io_placement foo.tcl` command.

rovinski, 2023-10-03:
@eder-matheus can PPL be adjusted to make `write_pin_placement` a separate command?

eder-matheus, 2023-10-03:
> @eder-matheus can PPL be adjusted to make `write_pin_placement` a separate command?

Done

maliberty, 2023-10-03:
Turn OR into an interactive layout editing system is an endless task that we have avoided.  This and #4068 are just the start once we open this door.  Is it possible to define improvements that would obviate the need to this manually?

oharboe, 2023-10-03:
@maliberty I just spoke with @louiic: he spoke of constraints to place macros. I think that coming up with some additional constraints is more idiomatic for OpenROAD's approach than to start down the path of GUI editing features...

That said, I think a write_io_placement and write_macro_placement Tcl commands is a good idea. For macros and io placement, this can be used to do some tinkering in text editors and also to speed up the flow by skipping macro and io placement...

maliberty, 2023-10-03:
Eder already did write_pin_placement in #4073 so I'll close this.  write_macro_placement wouldn't be hard and I'll revise #4068 to be just that.

mpl2 has some support for constraints that we could build on.

closed 2023-10-03 by maliberty