The-OpenROAD-Project/OpenROAD #4085: Improve RTLMP so that it can lay out mock-array in an array

oharboe, 2023-10-04:
### Description

unzip [rtlmp.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12808850/rtlmp.zip)

```
$ ./run-me-mock-array-asap7-base.sh 
OpenROAD v2.0-10574-gd04de2f32 
[deleted]
```

Placement result, not an array:

[image]

### Suggested Solution

Should result in an array.

AcKoucher, 2024-02-19:
@oharboe What criteria should induce the 8x8 array centering in the die? The predominant criteria used for the mechanism used by mpl2 is wirelength, however moving the array inside the die doesn't affect it at all due to the symmetry.

oharboe, 2024-02-19:
@AcKoucher @maliberty I dont know... We have discussed some sort of placement constraint language. Perhaps that is needed here to break the symmetry?

AcKoucher, 2024-02-26:
@oharboe With the changes in #4704 and #4705 we'll be able to centralize mock-array by setting the boundary weight parameter in HierRTLMP to 0. Would that be acceptable?

The default functionality of mpl2 is to push macros away from the center which is exactly the opposite behavior of what we'd want here.

oharboe, 2024-02-26:
@AcKoucher Sounds like a solid step forward.

@maliberty Any luck in getting.macro placement constraints articulated?

PR #4544 (2024-04-11): mpl2: add support for array of interconnected macros
Now that we have:
- #4614
- #4815
- #4206
- #4705

It's possible to resolve #4085 with the changes here and by setting the boundary penalty to 0.

Changes:
1. Adding another classification for single macro clusters based on interconnection during mixed leaves splitting.
2. Using this classification to merge macro clusters
3. Adding tight packing tiling computation for arrays of interconnected macros.
4. Add possibility to manually initialize sequence pair
5. Macro arrays have their sequence pair initialized to their respectful shape and SA o [... cut]
closed 2024-04-11 by maliberty