The-OpenROAD-Project/OpenROAD #4092: Error: cts.tcl, 110 couldn't execute "eqy": no such file or directory, asap7 jpeg

oharboe, 2023-10-05:
### Describe the bug

Run: `./build_openroad --local --latest`

```
[INFO RSZ-0094] Found 46 endpoints with setup violations.
[INFO RSZ-0040] Inserted 39 buffers.
[INFO RSZ-0033] No hold violations found.
Error: cts.tcl, 110 couldn't execute "eqy": no such file or directory
openroad> 
```

### Expected Behavior

Should run to completion without error

### Environment

```shell
Git commit: e70606743bb66db93f0a918975bff23455ad16c5
kernel: Linux 6.2.0-32-generic
os: Ubuntu 23.04 (Lunar Lobster)
cmake version 3.25.1
```

### To Reproduce

See above

oharboe, 2023-10-11:
@maliberty @vvbandeira I updated the testcase. No need for a giant download, happens with `make DESIGN_CONFIG=designs/nangate45/gcd/config.mk generate_abstract`

Works with `make DESIGN_CONFIG=designs/asap7/gcd/config.mk generate_abstract` though...

maliberty, 2023-11-06:
@vvbandeira please update

rovinski, 2023-11-06:
Both I and one of my students ran into this. I think one of the main problems is that there are default designs, e.g. nangate45/gcd, which have equivalence checking turned on. There either needs to be a check for whether `eqy` is actually installed, or `EQUIVALENCE_CHECK = 1` needs to be removed from designs.

maliberty, 2023-11-06:
@vvbandeira perhaps we can just call eqy from Jenkins (as the default)

kush-manchanda, 2024-01-15:
I had the same issue and had to follow the comment here https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1513#issuecomment-1741746622 to fix it. Thanks.

maliberty, 2024-01-15:
This should be resolved once https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1751 is merged

vvbandeira, 2024-01-15:
[#1751](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1751) was replaced by [#1754](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1754)

vvbandeira, 2024-01-16:
@kush-manchanda 
Could you try again with the latest code? Please also make sure to `source env.sh` before running the flow if you used `sudo ./setup.sh` to install the dependencies.

kush-manchanda, 2024-01-18:
@vvbandeira cloned it again and did the build locally. It worked fine.

closed 2024-01-18 by vvbandeira