The-OpenROAD-Project/OpenROAD #4094: pdn should error if a macro is failing to connect

maliberty, 2023-10-05:
### Describe the bug

If a macro is short enough to fall between two pdn strips then an error should be reported (or the algorithm improved to handle it).

### Expected Behavior

A PDN error rather than a PSM one.

### Environment

```shell
See below
```

### To Reproduce

ng45/bp when run with 
ORFS 9d638a19b69c167b61479ec311b40b87d2107cad
OR e49038b1b3242f838daf22a52797de3137883675

(eg Jenkins OpenROAD-flow-scripts-Nightly-Public/3090)

### Relevant log output

```shell
From https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/3090/execution/node/842/log/

[WARNING PSM-0038] Unconnected PDN node on net VDD at location (27.485um, 322.455um), layer: metal5.
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
```

### Screenshots

[image]

rovinski, 2023-10-06:
I will throw out there that it can be useful to ignore these errors when prototyping. It would be nice to have an option to ignore a hypothetical error during the PDN stage or have it be automatically fixed with a healing strap or similar.

maliberty, 2026-03-24:
This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 21 days if no further activity occurs. Remove the Stale label or comment to keep it open.