The-OpenROAD-Project/OpenROAD #4104: PPL - Less pin positions available than expected when using define_pin_shape_pattern

lustefan97, 2023-10-06:
### Describe the bug

This is a follow up to #4061, I know that it should have been fixed by #4071 but it doesn't seems to work even after pulling down the lastest commits and re-building OpenROAD.

I have a design whose core area is a bit more than 320um x 320um, my minimum pitch is 0.32 for the x/y axis so I expect to have 1000*1000 = 1 000 000 pin position available but OpenROAD tells me that it only has 13 568 pins positions available.

Just as info : I use OpenROAD-flow-script. To update OpenROAD I have gone in the OpenROAD-flow-scripts/tools/OpenROAD, used `git checkout master` then `git pull`, followed then by `cd build` and `make`. I even tried `git submodule update --remote` and `git submodule update --remote --recursive`, to see if it came from the use of submodule.
Building OpenROAD after pulling down the latest commits didn't seemed to fix the issue, I hope it isn't just a mishandling on my part...

### Expected Behavior

Pin placement allowed due to sufficient pin locations.

### Environment

```shell
Git commit: 672579366902e0f3f1561320096912cf84d688b4
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)

(Note that I just did another commit, but the build should be from the last commit from October 5th)
```

### To Reproduce

[TestCase.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12834507/TestCase.zip)

Here should be every needed file to reproduce the issue. Create a TestCase directory in _OpenROAD-flow-script/tools/OpenROAD/test/_ and extract every file there. Open the gui with `openroad -gui` and enter `source "ldpc4x4_asap7_bottom.tcl"` in the TCL command box.

### Additional Context

I also noticed that the RAM usage could go as high as 15GB with the following test case (with 16GB allowed at maximum for the Linux VM), I don't know if it is normal or not.

eder-matheus, 2023-10-06:
@lustefan97 I'm having this error when running your attached test case during placement:
```
[ERROR GPL-0302] Use a higher -density or re-floorplan with a larger core area.
Given target density: 0.60
Suggested target density: 0.61
```

Could you add the log file where the number of available positions is reported? Do you see it in an error message or just in an INFO message?

lustefan97, 2023-10-07:
@eder-matheus I don't think I even got to global placement, I was stuck in the (random) IO pin placement just before that.

I feel like it is an error on my part, where I failed to update OpenROAD... Just to be sure, in order to properly update OpenROAD, I just have to go into _OpenROAD-flow-scripts/tools/OpenROAD/_, use `git checkout master` then `git pull`and finally build OpenROAD again by going into _OpenROAD/build/_ and using `make` to compile everything back again.

I know that my files are up-to-date with the master branch, but it seems that I failed to build OpenROAD with the recent changes....

lustefan97, 2023-10-07:
Yes indeed, it was a bad manipulation on my part, I do have the same error as you do for the global placement now... Sorry for the inconvenience

closed 2023-10-07 by lustefan97

eder-matheus, 2023-10-07:
@lustefan97 No problem, you've raised an essential missing in the PPL tool. It wasn't reporting the number of positions from the top layer grid; my PR is covering it now.

Also, when using the OpenROAD-flow-scripts, you can use the build_openroad.sh script to get the latest updates in the submodules. So, if you want to build ORFS with the latest commits from the OpenROAD repository, you can run `./build_openroad.sh --latest`. This will update all submodules to the latest commit. Also, you can use the flag `--local` to build everything in your machine instead of docker.