The-OpenROAD-Project/OpenROAD #4108: Global routing fails to route, even if there is a lot of space

oharboe, 2023-10-07:
### Description

unzip https://drive.google.com/file/d/1BFUHnN-5qJI7WUqug3ehs6fxSkNoZCqJ/view?usp=sharing

Run:

```
$ ./run-me-mock-aluregisterfile-asap7-base.sh 
OpenROAD v2.0-10613-g75796f6ec 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GRT-0020] Min routing layer: M2
[INFO GRT-0021] Max routing layer: M7
[INFO GRT-0022] Global adjustment: 0%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer M1      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M2      Track-Pitch = 0.0390  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M3      Track-Pitch = 0.0360  line-2-Via Pitch: 0.0360
[INFO GRT-0088] Layer M4      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M5      Track-Pitch = 0.0480  line-2-Via Pitch: 0.0480
[INFO GRT-0088] Layer M6      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0088] Layer M7      Track-Pitch = 0.0640  line-2-Via Pitch: 0.0640
[INFO GRT-0019] Found 892 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 31
[INFO GRT-0003] Macros: 4
[INFO GRT-0004] Blockages: 2496

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
M1         Vertical            0             0          0.00%
M2         Horizontal   15151320       4140683          72.67%
M3         Vertical     17502480       5557969          68.24%
M4         Horizontal   12835800       4107283          68.00%
M5         Vertical     12835800       3683262          71.30%
M6         Horizontal    9335520       2665179          71.45%
M7         Vertical      9335520       3282814          64.84%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 55329
[INFO GRT-0198] Via related Steiner nodes: 2322
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 92187
[INFO GRT-0112] Final usage 3D: 2476739
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/asap7/mock-aluregisterfile/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 27 GRT-0119
openroad> 

```

oharboe, 2023-10-07:
Just for information, this is the mock-aluregisterfile design where I try to break out the registerfile as a macro and split it into 4 equally sized multiported SRAMs that are combined at the top level.

This makes running times more managable:

```
$ make elapsed
[INFO][FLOW] Using platform directory /home/oyvind/OpenROAD-flow-scripts/flow/platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
User PVT selection: BC
[INFO][FLOW] Invoked hierarchical flow.
Block registers_32x64 needs to be hardened.
./logs/asap7/mock-aluregisterfile_registers_32x64/base/
Log                       Elapsed seconds
1_1_yosys                         48
2_1_floorplan                      8
2_2_floorplan_io                   1
2_3_floorplan_tdms                 1
2_4_floorplan_macro                1
2_5_floorplan_tapcell              1
2_6_floorplan_pdn                  1
3_1_place_gp_skip_io               9
3_2_place_iop                      1
3_3_place_gp                      39
3_4_place_resized                 28
3_5_place_dp                      18
4_1_cts                           34
4_2_cts_fillcell                   1
5_1_grt                           21
5_2_route                        917
6_1_merge                          4
6_report                          49
./logs/asap7/mock-aluregisterfile/base
Log                       Elapsed seconds
1_1_yosys                          5
2_1_floorplan                      1
2_4_floorplan_macro                1
2_5_floorplan_tapcell              1
2_6_floorplan_pdn                 13
3_1_place_gp_skip_io              40
3_2_place_iop                      2
3_3_place_gp                      57
3_4_place_resized                  7
3_5_place_dp                       8
4_1_cts                           11
4_2_cts_fillcell                   6
5_1_grt.tmp                       94
```

maliberty, 2023-11-02:
@eder-matheus any update?

closed 2024-02-20 by eder-matheus