The-OpenROAD-Project/OpenROAD #4124: [ERROR PDN-0233] Failed to generate full power grid.

oharboe, 2023-10-12:
### Describe the bug

unzip [pdn-error.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12880471/pdn-error.zip)

```
$ ./run-me-aes_cipher_top-asap7-base.sh 
OpenROAD v2.0-10646-g0dff30da2 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: top
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u0/r0
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u0/u0
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u0/u1
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u0/u2
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - u0/u3
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us00
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us01
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us02
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us03
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us10
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us11
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us12
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us13
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us20
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us21
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us22
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us23
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us30
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us31
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us32
[INFO PDN-0001] Inserting grid: CORE_macro_grid_1 - us33
[WARNING PDN-0232] CORE_macro_grid_1 - u0/u0 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - u0/u1 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - u0/u2 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - u0/u3 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us00 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us01 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us02 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us03 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us10 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us11 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us12 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us13 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us20 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us21 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us22 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us23 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us30 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us31 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us32 does not contain any shapes or vias.
[WARNING PDN-0232] CORE_macro_grid_1 - us33 does not contain any shapes or vias.
[ERROR PDN-0233] Failed to generate full power grid.
Error: pdn.tcl, 5 PDN-0233
openroad> 
```

### Expected Behavior

No error

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 6.2.0-34-generic
os: Ubuntu 23.04 (Lunar Lobster)
cmake version 3.25.1
```

### To Reproduce

See above

oharboe, 2023-10-12:
Still fails with OpenROAD v2.0-10652-ged4dea565

oharboe, 2023-10-12:
Perhaps this is a problem with macro placement?

Including a standalone test for macro placement for convenience...

[macro-place.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12880806/macro-place.zip)

```$ ./run-me-aes_cipher_top-asap7-base.sh 
OpenROAD v2.0-10652-ged4dea565 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
HierRTLMP Flow enabled...
Call Macro Placer  -halo_width 5 -report_directory build//objects/asap7/aes_cipher_top/base/rtlmp
Floorplan Outline: (0.0, 0.0) (202.022, 202.022),  Core Outline: (2.052, 2.16) (200.016, 199.8)
Traversed logical hierarchy
	Number of std cell instances : 4226
	Area of std cell instances : 517.59
	Number of macros : 21
	Area of macros : 7310.76
	Area of macros with halos : 17195.82
	Total area : 7828.35
	Design Utilization : 0.20
	Core Utilization: 0.02
	Manufacturing Grid: 1

[INFO MPL-0024] Perform Clustering..
[INFO MPL-0028] Perform Multilevel macro placement...
[INFO MPL-0031] [MultiLevelMacroPlacement] Working on children of cluster: root, Outline 2.052, 2.16  197.964, 197.64
[INFO MPL-0031] [MultiLevelMacroPlacement] Working on children of cluster: u0, Outline 2.052, 2.16  29.013, 197.64
Align macros within the cluster u0
[INFO MPL-0031] [MultiLevelMacroPlacement] Working on children of cluster: (root)_glue_logic, Outline 31.065, 2.16  168.951, 197.64
Align macros within the cluster (root)_glue_logic
Align macros within the cluster root
number of updated macros : 21
number of macros in HardMacroCluster : 21
Delete buffers for RTLMP flow...
[INFO RSZ-0026] Removed 0 buffers.
openroad> 
```

maliberty, 2023-10-12:
The macros are fully blocked on M4 & M5 so there is no way to connect them with
```
define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro -orient {R0 R180 MX MY} -cells {.*}
add_pdn_connect -grid {CORE_macro_grid_1} -layers {M4 M5}
```

closed 2023-10-12 by maliberty

b224hisl, 2023-12-04:
> The macros are fully blocked on M4 & M5 so there is no way to connect them with
> 
> ```
> define_pdn_grid -name {CORE_macro_grid_1} -voltage_domains {CORE} -macro -orient {R0 R180 MX MY} -cells {.*}
> add_pdn_connect -grid {CORE_macro_grid_1} -layers {M4 M5}
> ```

So I'm not quite understand, could you please give an example to illustrate how to generate pdn successfully in a design with macros?

maliberty, 2023-12-05:
Your macros have blockages above the layers you are trying to connect to the grid on (m4/m5).  How were your macros generated?

oharboe, 2023-12-05:
This issue is closed, are these comments for a different issue?

 I simply used the wrong PDN script for the macro. In ASAP7 there are two PDN scripts, one for macros and one for top level. If you use the top level PDN script for a.macro, then you get the failure above.

maliberty, 2023-12-05:
After I closed it a further question was asked.

Lukemagik, 2024-04-25:
@b224hisl Did you manage to find a way to correctly generate the macro? I'm getting that error too.

[image]