The-OpenROAD-Project/OpenROAD #4132: Openroad interferes with company VPN

msingh9, 2023-10-14:
### Describe the bug

It is very vague issue I am reporting. Openroad is installed on Ubuntu machine by company's IT. Since the installation, I am the only one who is running openroad in a loop. It has brought down our VPN twice during last 24 hours. This is strange coincidence, that IT  thinks it is because of openroad SW.  They have never eperienced this kind of intermittent outage before. It happened earlier too where network issue was narrowed down to my local linux desktop which has openroad installed and running.

I know I am not providing concrete information, but if anyone has experienced such thing or have a clue, please guide us. I would like to continue using openroad, but the network issue caused IT shut down that machine.

### Expected Behavior

No network issues

### Environment

```shell
Ubuntu machine
```

### To Reproduce

I don't have a test case.

maliberty, 2023-10-14:
I have no idea and can't conceive of a mechanism by which this could happen.

maliberty, 2023-10-17:
Nothing actionable.

closed 2023-10-17 by maliberty