The-OpenROAD-Project/OpenROAD #4142: generate_abstract crash

oharboe, 2023-10-17:
### Describe the bug

unzip [generate_abstract.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/12929310/generate_abstract.zip)

```
$ ./run-me-CSRFile-asap7-base.sh 
OpenROAD v2.0-10698-g5f38dd68b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Starting generation of abstract views
Signal 11 received
Stack trace:
 0# 0x000055ADCFEC5723 in openroad
 1# 0x00007FB11F63C460 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::ClkSkews::findClkDelays(sta::Clock const*, sta::MinMaxValues<float> (&) [2][2]) in openroad
 3# sta::MakeTimingModel::findClkInsertionDelays() in openroad
 4# sta::MakeTimingModel::makeTimingModel() in openroad
 5# sta::makeTimingModel(char const*, char const*, char const*, sta::Corner const*, sta::Sta*) in openroad
 6# sta::Sta::writeTimingModel(char const*, char const*, char const*, sta::Corner const*) in openroad
 7# 0x000055ADD00A6806 in openroad
 8# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# 0x00007FB123DB3C43 in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
11# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
12# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
13# ord::tclAppInit(Tcl_Interp*) in openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in openroad
16# 0x00007FB11F623A90 in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in openroad
./run-me-CSRFile-asap7-base.sh: line 7: 563863 Segmentation fault      (core dumped) openroad -no_init ${SCRIPTS_DIR}/generate_abstract.tcl
```

### Expected Behavior

No crash

### Environment

```shell
OpenROAD v2.0-10698-g5f38dd68b
```

### To Reproduce

See above

maliberty, 2023-10-17:
I'll pass along the crash but the trigger seems to be that you have a signal io_ungated_clock that is used to control FF (eg _14268_/CLK) but is not a clock in your SDC.

maliberty, 2023-10-17:
check_setup shows:
```
Warning: There are 65 unclocked register/latch pins.
Warning: There are 232 unconstrained endpoints.
```

PR #4147 (2023-10-18): Update sta
Fixes #4142
commit e9734a52 (2023-10-18): Update sta

Fixes #4142
closed 2023-10-18 by maliberty