The-OpenROAD-Project/OpenROAD #4145: MPL2 is not properly snapping macros when using Nangate45 designs.

AcKoucher, 2023-10-17:
### Describe the bug

Instead of aligning the pins according to the track-grid, the pins are getting slightly misplaced.

### Expected Behavior

Pins should be centralized according to the track-grid independently of the technology used.

### Environment

```shell
Head of Master
```

### To Reproduce

Run mpl2 in a nangate45 design with macro instances.

### Screenshots

nangate45/bp_be_top - fakeram45_512x64 - some pins in metal3.
[image]

AcKoucher, 2023-10-26:
After taking a look at some of the pins of the macro `be_mmu/dcache/tag_mem.macro_mem.mem` in `nangate45/bp_be_top`, I realized it has a arrangement of pins which apparently makes impossible for mpl2 to snap it:

Left boundary pins in metal3 - `nangate45/bp_be_top`, macro `be_mmu/dcache/tag_mem.macro_mem.mem`
[image]

This also happens with some macros in nangate45/tinyRocket, e. g. `dcache/data.data_arrays_0.data_arrays_0_ext.mem`.

On the other hand, some macros which don't have this kind of arrangement (or which have pins that are much larger than the trackgrid pitch) aren't being correctly snapped as well. I'm investigating.

maliberty, 2023-10-26:
It would be nice to find the position that snaps as many pins as possible as in the image above none are snapped.

What is the name of the block shown above?  Is it a fake ram?

AcKoucher, 2023-10-26:
In the image: bp_be_top, and the macro is be_mmu/dcache/tag_mem.macro_mem.mem. It's a fake ram.

AcKoucher, 2023-10-26:
However, in the case of the image, if we snap the lower and upper pins, the middle one (as many others) will be completely out of the track...

maliberty, 2023-10-26:
Probably we should just fix the RAM to have its pins arranged so that snapping is possible.  Fakeram has been through various iterations of improvements.

maliberty, 2023-10-26:
See https://github.com/ABKGroup/FakeRAM2.0

rovinski, 2023-10-28:
At some point it might be prudent to create an IP checker which can more thoroughly check for bad implementations such as off-grid pins, wrong direction routing, etc.

maliberty, 2023-10-28:
Those could be enhancements to check_placement

closed 2023-11-14 by AcKoucher