The-OpenROAD-Project/OpenROAD #4148: global route crash

oharboe, 2023-10-18:
### Describe the bug

This is from MegaBoom:

```
./orfs make DESIGN_NAME=RegisterFileSynthesizable_1 DESIGN_CONFIG=block.mk
VERILOG_FILES="rtl/RegisterFileSynthesizable_1.sv\ rtl/regfile_128x64.sv" CORE_UTILIZATION=20 PLACE_DENSITY=0.40
```

```
OpenROAD v2.0-10698-g5f38dd68b 
[deleted]
[INFO GRT-0103] Extra Run for hard benchmark.
/usr/include/c++/12/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = int; _Alloc = std::allocator<int>; reference = int&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Signal 6 received
Stack trace:
 0# 0x00005563040E3323 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F3F3E03C460 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007F3F3E4E21CF in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# grt::FastRouteCore::copyGrids(grt::TreeNode const*, int, int, grt::TreeEdge const*, int, std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# grt::FastRouteCore::updateRouteType1(int, grt::TreeNode const*, int, int, int, int, int, grt::TreeEdge*, int, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 8# grt::FastRouteCore::mazeRouteMSMD(int, int, float, int, int, bool, int, float, int, int, int, float&) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 9# grt::FastRouteCore::run() in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
10# grt::GlobalRouter::findRouting(std::vector<grt::Net*, std::allocator<grt::Net*> >&, int, int) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
11# grt::GlobalRouter::globalRoute(bool, bool, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x000055630476123C in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
14# 0x00007F3F4295FC43 in /lib/x86_64-linux-gnu/libtcl8.6.so
15# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
16# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
17# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
18# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
19# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
20# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
21# 0x00007F3F3E023A90 in /lib/x86_64-linux-gnu/libc.so.6
22# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
23# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 6
Elapsed time: 59:28.99[h:]min:sec. CPU time: user 3567.78 sys 0.95 (99%). Peak memory: 2846924KB.
```

### Expected Behavior

no crash

### Environment

```shell
OpenROAD v2.0-10698-g5f38dd68b
```

### To Reproduce

See above

oharboe, 2023-10-21:
@maliberty @luis201420 I'm still running deltaDebug.py and it is making progress. However, I think it could take a week to whittle down to something quicker than 2hrs. So far a successful run takes hours to time out and deltaDebug.py is failing to eliminate large swathes early. 82 iterations * 2 hours so far... Fortunately it is winter in Norway and the heat from my machine in my home-office is not wasted :-)

luis201420, 2023-10-25:
Hi @oharboe,
Could you try to run the design again? The bug was fixed and the solution is already in the master branch.

oharboe, 2023-11-01:
@luis201420 deltaDebug.py is still running, 633 iterations so far...

oharboe, 2023-11-01:
If I see the problem again, I'll file a new issue.

closed 2023-11-01 by oharboe