The-OpenROAD-Project/OpenROAD #4185: Clock tree is not balanced because of clock gating

mo-hosni, 2023-10-29:
### Describe the bug

I don't think this is a bug, it is just a limitation.
The issue is:
I am implementing a design that has part of the clock tree gated. My issue is that in CTS, I can see that there are two clock trees, one for the master clock and one for the gated clock. I can see that in the log too:
```
[INFO CTS-0010]  Clock net "mprj.CLK" has 10518 sinks.
[INFO CTS-0010]  Clock net "mprj.clk_gated" has 1641 sinks.
```
The issue is that any path from a FF in one of the clock trees to a FF in the other clock tree has a large skew hence setup and hold violations.
The clock gating is done using a single latch cell. When I remove this cell, the skew become minimal and everything is fine.
So, is there a way that OpenROAD could ignore this cell in CTS and just add it to its leaf so that we have 1 clock tree that is balanced?
I tried adding `-balance_levels` to the  `clock_tree_synthesis` but it didn't get balanced.

### Expected Behavior

Balanced clock trees across all sinks I guess.

### OpenROAD Environment

```shell
OpenROAD version: 0a584d123190322b0725d5440c2c486d91d3afd8
```

### OpenLane Environment

```shell
n/a
```

### To Reproduce

[reproducible.zip.partaa.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13198349/reproducible.zip.partaa.zip)
[reproducible.zip.partab.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/13198351/reproducible.zip.partab.zip)

maliberty, 2023-10-29:
balance_levels should ensure there are the same number of buffers in all paths but it doesn't ensure that the delays match.  Do you at least see that much?

mo-hosni, 2023-10-29:
@maliberty I can see it did ensure that all paths have the same number of buffers, yes. But the skew is still more than 2ns.

Now I looked at 1 of the violating timing paths before and after adding clock gating:

Without clock gating:
The launching flipflop clock had 11 buffers, `3.0 ns` delay
The receiving flipflop clock had 10 buffers, `2.2 ns` delay
the skew was `0.8 ns` and I didn't use balance_levels as the resizer did fix all hold and setup violations.

With the clock gating latch:
The launching flipflop clock had 14 buffers + latch, `5.0 ns` delay
The receiving flipflop clock  had 14 buffers, `2.9 ns` delay
the skew now is: `2.1 ns` and I used balance_levels to add some delay for the receiving flipflop clock.

balance_levels did help but the the delay wasn't matched, yes.

maliberty, 2023-10-29:
Reporting the skew w/o balance_levels is of no interest as that's known to not work well with clock gating.  Reporting them with balance_levels would be more helpful.

mo-hosni, 2023-10-29:
Okay, I will can using balance_levels in both cases. But, is there a solution using OpenROAD for balancing clock trees that has clock gating?

maliberty, 2023-10-30:
@precisionmoon : " It's kind of urgent for us" from @donn 

I also tried adding -obstruction_aware but the results are still pretty bad:
[image]

maliberty, 2023-10-30:
If I manually change mprj.cpu_clk_gate from sky130_fd_sc_hd__dlclkp_1 to sky130_fd_sc_hd__dlclkp_4 then I get:

[image]

maliberty, 2023-10-31:
@donn confirms that the manual workaround resolves the issue for their immediate tapeout.  However we should still fix this when you can.

azwefabless, 2023-11-01:
@maliberty Best status I can gather from a quick download from the team: The Hold timing issue is improved to the point of it being possible to get a netlist by the workaround but at the cost of more than doubling total chip pwr consumption and still requires the addition of wait states to read the flash due to the router stopping working on resolving setup issues. I think we likely need further help here. This is still a significant issues for us,

maliberty, 2023-11-01:
This is being discussed in slack so I'll respond there