The-OpenROAD-Project/OpenROAD #4186: Detailed Routing should not be limited to 64 iterations

Peter-Herrmann, 2023-10-29:
### Description

Detailed Routing is currently limited to 64 iterations by limiting the max input on `DRT_OPT_ITERS`. This limit is problematically low, I have several designs that fail because the max iterations are hit when the number of violations is still dropping (I just had a run where the violation count dropped from 2 to 1 on the 64th iteration). Later optimization passes take a fraction of a second, and the fact that the maximum possible value is the default setting for OpenLane when calling DRT suggests that this is generally not enough.

### Suggested Solution

The maximum value for DRT_OPT_ITERS should be as high as possible. [This check](https://github.com/The-OpenROAD-Project/OpenROAD/blob/0074941fb9569768fff3be72e2b6b01d5b142196/src/drt/src/TritonRoute.tcl#L120) should have its cap raised or the check should be removed if there is no affirmative reason to have it.

### Additional Context

I'm not able to find a reason why this value is capped at 64 when looking through the code base, it appears arbitrary.

OpenRoad drt checks if the value passed to `droute_end_iter` is in the range [1,64] and [rejects other values](https://github.com/The-OpenROAD-Project/OpenROAD/blob/0074941fb9569768fff3be72e2b6b01d5b142196/src/drt/src/TritonRoute.tcl#L120).

The value of `droute_end_iter` is [passed to tritonroute](https://github.com/The-OpenROAD-Project/OpenROAD/blob/0074941fb9569768fff3be72e2b6b01d5b142196/src/drt/src/TritonRoute.tcl#L194) through [this detailed route cmd](https://github.com/The-OpenROAD-Project/OpenROAD/blob/0074941fb9569768fff3be72e2b6b01d5b142196/src/drt/src/TritonRoute.i#L105), which [parses it into `END_ITERATION`](https://github.com/The-OpenROAD-Project/TritonRoute/blob/758cdac457069002f7b1d96192dfc35929216799/src/main.cpp#L72)

Tritonroute does not seem to have any apparent limitation above 64 iterations, ***in fact the [default value](https://github.com/The-OpenROAD-Project/TritonRoute/blob/758cdac457069002f7b1d96192dfc35929216799/src/global.cpp#L79) of  `END_ITERATION` in tritonroute is actually 80***, but the wrapper scripts will reject any value above 64.

rovinski, 2023-11-02:
There's actually a simple reason for the limit, which is that each iteration is a different "strategy" and there are only 64 different strategies coded into DRT:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/0074941fb9569768fff3be72e2b6b01d5b142196/src/drt/src/dr/FlexDR.cpp#L958-L1031

After 64 iterations, there are no more strategies to try. One could add more strategies, but there are severe diminishing returns for adding more strategies. If there are still violations after 64 iterations, then it's more likely that there are algorithmic improvements that need to be made to the router (or that the design's routability is infeasible).

There's lots of room for improvement in the router, but it's a complex and challenging issue.

If you have a specific test case that you are able to share, someone might be able to see if there's a systemic issue with the router (e.g. a bug, poor support for a specific DRC rule, unexpected corner case etc.) or if there's no clear problem other than the design's density is too high.

maliberty, 2023-11-02:
Note that the standalone TritonRoute repo is an archive and all current development happens in the drt module of OR.

maliberty, 2023-11-06:
I think this has been answered.  There is no plan for more iterations.

closed 2023-11-06 by maliberty